raspiblitz/raspiblitz #3182: Make the WebUI + API optional

openoms, 2022-06-22:
Ran some tests with CLN which is still incomplete partly due to the `cln-grpc` interface being only experimental at it's first v0.1 release and @fusion44 soing most of the pioneering work himself with the API: https://github.com/fusion44/blitz_api/issues/125
While working suboptimally the API generates a significant load even with minimal services running:
[image]

I think after the recovery / initial setup the webUI+API should be optional with a choice in the ssh menu SERVICES to deactivate (even as first thing). For security and hardware load reasons this will be still desirable even when everything will work.
Could be on by default for recovery / initial setup in the release image, but the build_sdcard.sh should have an option to skip building it.

fusion44, 2022-06-22:
I've opened an issue for these performance problems in the API repo. 

Roughly, how many invoices and payments does this CLN node have?

openoms, 2022-06-22:
@fusion44 not many. Under a hundred each, with only 5 public channels currently.

openoms, 2022-06-24:
Another tough question is if the api really needs to run as `root` ?

It should be it's own separate, non-sudo user.
To access the LND creds it can be added to the `lndadmin` group (like `bos`).
To access the `lightning-rpc` it needs be added to the `bitcoin` group (like `lnbits`).

rootzoll, 2022-07-27:
OK - the sd card can be build without WebUI & API (no fatpack with option: `-f 0`) and also in the SSH menu under SETTINGS the WebUI & API can be switched off. Closing issue for v1.8.0 release.

closed 2022-07-27 by rootzoll