raspiblitz/raspiblitz #3184: Feature Request: Web UI Console

Surenic, 2022-06-23:
It would be pretty neat to have a console inside the new web UI like for example the Homebridge UI (image below) has.

That way you could fully control the node right in the browser. No need to extra log in by Terminal or any other software like PuTTY

[image]

openoms, 2022-06-24:
See this issue: https://github.com/rootzoll/raspiblitz/issues/2767#issue-1072605211

exposing your admin access to the web is not without tradeoffs though

rootzoll, 2022-06-28:
Closing as duplicate - https://github.com/rootzoll/raspiblitz/issues/2767#issue-1072605211

closed 2022-06-28 by rootzoll