raspiblitz/raspiblitz #3185: Sync of blockchain was fast but sync of lightning-wallet is taking days

stn021, 2022-06-23:
Hello,

I set up a new raspiblitz and synced the blockchain by copy over lan. That took a few hours and completed as expected. Now it reports "Blockchain Progress" as 100%.

But the "Lightning progress" has been going on for days and still is only 65%.

I thought that selfsync of the blockchain can take days but now what is going on with the lightning-wallet? I thought that would sync in a very short time.

Did I make some mistake there or what can I do to speed this up?

THX, regards, stn

openoms, 2022-06-24:
ctrl+c
type:
`menu`
see the logs in the SYSTEM menu
or tyoe:
`debug`

stn021, 2022-06-24:
OK, found:

- bitcoin seem OK to me, log shows a few connected peers, progress is still "100%"
- tor seems OK, log shows 5-10KB/sec

- CL-Log show only one line, related to the summary-plugin:
2022-06-24T02:47:34.722Z UNUSUAL plugin-summary.py: [PriceThread] Cannot allocate write+execute memory for ffi.callback(). You might be running on a system that prevents this. For more information, see https://cffi.readthedocs.io/en/latest/using.html#callbacks

In the logs I see no error-messages that I recognize.

debug says: 

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.2
chainnetwork: bitcoin / main
 10:34:17 up 5 min,  2 users,  load average: 3.13, 1.90, 0.82

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-06-24 10:30:37 BST; 3min 40s ago
    Process: 6150 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6165 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6167 ExecStart=/usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6201 (bitcoind)
      Tasks: 18 (limit: 9354)
        CPU: 1min 24.580s
     CGroup: /system.slice/bitcoind.service
             └─6201 /usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Jun 24 10:30:00 raspberrypi systemd[1]: Starting Bitcoin daemon...
Jun 24 10:30:37 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Fri 2022-01-28 03:21:52 GMT, ends at Fri 2022-06-24 10:34:17 BST. --
Jun 24 10:30:00 raspberrypi systemd[1]: Starting Bitcoin daemon...
Jun 24 10:30:37 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-06-24T09:30:37Z 0 block-relay-only anchors will be tried for connections.
2022-06-24T09:30:37Z init message: Starting network threads…
2022-06-24T09:30:37Z DNS seeding disabled
2022-06-24T09:30:37Z init message: Done loading
2022-06-24T09:30:37Z addcon thread start
2022-06-24T09:30:37Z msghand thread start
2022-06-24T09:30:37Z net thread start
2022-06-24T09:30:37Z opencon thread start
2022-06-24T09:30:41Z Imported mempool transactions from disk: 2525 succeeded, 0 failed, 0 expired, 0 already there, 0 waiting for initial broadcast
2022-06-24T09:30:41Z loadblk thread exit
2022-06-24T09:31:16Z New outbound peer connected: version: 70016, blocks=742118, peer=0 (outbound-full-relay)
2022-06-24T09:31:36Z New outbound peer connected: version: 70016, blocks=742118, peer=4 (outbound-full-relay)
2022-06-24T09:32:06Z New outbound peer connected: version: 70016, blocks=742118, peer=5 (outbound-full-relay)
2022-06-24T09:32:13Z New outbound peer connected: version: 70016, blocks=742118, peer=6 (outbound-full-relay)
2022-06-24T09:32:21Z New outbound peer connected: version: 70016, blocks=742118, peer=7 (outbound-full-relay)
2022-06-24T09:33:01Z New outbound peer connected: version: 70016, blocks=742118, peer=9 (outbound-full-relay)
2022-06-24T09:33:09Z New outbound peer connected: version: 70016, blocks=742118, peer=10 (outbound-full-relay)
2022-06-24T09:33:41Z New outbound peer connected: version: 70016, blocks=742118, peer=11 (outbound-full-relay)
2022-06-24T09:33:52Z New outbound peer connected: version: 70016, blocks=742118, peer=13 (block-relay-only)
[... 198 lines cut ...]
btc_default_version="v22.0.0"
btc_default_error_short=""
btc_default_sync_initial_done="1"
btc_default_blocks_behind="0"
btc_default_error_full=""
btc_default_mempool_transactions="2980"
btc_default_online="1"
btc_default_blocks_headers="742118"
btc_default_running="1"
btc_default_peers="12"
btc_default_ready="1"
btc_default_sync_percentage="100.00"
btc_default_blocks_verified="742118"
btc_default_sync_progress="0.9999960423830677"
btc_default_sync_initialblockdownload="0"
btc_default_synced="1"
btc_default_activated="1"
btc_default_port="8333"

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

openoms, 2022-06-25:
There is some  installation error with your summary plugin. If a restart does not help reflash the Sdcard.

stn021, 2022-06-25:
As I wrote in my previous post, first line: 'The SD-card has been reflashed once already, then recover, then 'patch'.'
That did not help.
What else can I check?

openoms, 2022-06-25:
`Cannot allocate write+execute memory for ffi.callback`
Could it be an SDcard error?
Consider using a different Endurance type card.

openoms, 2022-06-25:
Can also try a restart (after which the summary plugin doesn't run)
and try to delete plugins directory and  reinstall the summary plugin

stn021, 2022-06-25:
For this raspiblitz I actually used two different cards. 

Flashed a new one, then swapped. Meaning, I did not reuse the same one. So even if one has an error that does not explain why lightning takes so long to set up its database.
Also I use Sandisk Ultra Plus. I have used them with different applications of the raspberry pi, for quite a while now and have compared the with other SD-cards. They are very enduring, also almost new.

openoms, 2022-06-25:
```
sudo rm -rf  /home/bitcoin/cl-plugins-available/plugins
config.scripts/cl-plugin.summary.sh on mainnet
sudo systemctl restart lightningd
```

stn021, 2022-06-25:
FMI: what exactly happens when the lightning-wallet syncs after installation.
For the btc-wallet that is obvious, if downloads the blockchain. That takes a while, understandably.

But there is no blockchain for lightning. A new lightning-wallet should contain no external data at all. So what does it download and sync?

stn021, 2022-06-25:
OK, I will reset the plugin.

openoms, 2022-06-25:
scans the bitcoin blocks and downloads the lightning network graph

stn021, 2022-06-28:
> 

> `Cannot allocate write+execute memory for ffi.callback` Could it be an SDcard error? Consider using a different Endurance type card.

FYI: here this happens on every node, every time the cln-summary-plugin is called. It is not related to the issue in this thread, it just appears at the same time.

openoms, 2022-06-28:
Please look at the logs to see what happening.
Use `debug` or the LND logs from the SYSTEM menu.

stn021, 2022-06-28:
Here it is. I do not see anything special.
Even worse: I just restarted the node. Now it is again at Lightning Progress : 94.49%. 
Before the reboot it was at 96%.

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.2
chainnetwork: bitcoin / main
 08:14:43 up 16 min,  2 users,  load average: 2.54, 2.59, 1.84

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-06-28 07:59:35 BST; 15min ago
    Process: 5472 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 5480 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 5484 ExecStart=/usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 5529 (bitcoind)
      Tasks: 18 (limit: 9354)
        CPU: 3min 16.628s
     CGroup: /system.slice/bitcoind.service
             └─5529 /usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Jun 28 07:59:01 raspiblitz-81 systemd[1]: Starting Bitcoin daemon...
Jun 28 07:59:35 raspiblitz-81 systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Fri 2022-01-28 03:21:52 GMT, ends at Tue 2022-06-28 08:14:43 BST. --
Jun 28 07:59:01 raspiblitz-81 systemd[1]: Starting Bitcoin daemon...
Jun 28 07:59:35 raspiblitz-81 systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-06-28T06:59:35Z DNS seeding disabled
2022-06-28T06:59:35Z net thread start
2022-06-28T06:59:35Z init message: Done loading
2022-06-28T06:59:35Z msghand thread start
2022-06-28T06:59:35Z opencon thread start
2022-06-28T06:59:35Z addcon thread start
2022-06-28T06:59:43Z Imported mempool transactions from disk: 1533 succeeded, 0 failed, 0 expired, 0 already there, 0 waiting for initial broadcast
2022-06-28T06:59:43Z loadblk thread exit
2022-06-28T06:59:53Z New outbound peer connected: version: 70016, blocks=742675, peer=0 (outbound-full-relay)
2022-06-28T07:00:05Z New outbound peer connected: version: 70016, blocks=742675, peer=1 (outbound-full-relay)
2022-06-28T07:00:35Z New outbound peer connected: version: 70016, blocks=742675, peer=2 (outbound-full-relay)
2022-06-28T07:00:50Z New outbound peer connected: version: 70016, blocks=742675, peer=3 (outbound-full-relay)
2022-06-28T07:01:09Z New outbound peer connected: version: 70016, blocks=742675, peer=4 (outbound-full-relay)
2022-06-28T07:02:00Z New outbound peer connected: version: 70016, blocks=742675, peer=5 (outbound-full-relay)
2022-06-28T07:03:14Z New outbound peer connected: version: 70016, blocks=742675, peer=8 (outbound-full-relay)
2022-06-28T07:03:32Z New outbound peer connected: version: 70016, blocks=742675, peer=9 (outbound-full-relay)
2022-06-28T07:03:41Z New outbound peer connected: version: 70016, blocks=742675, peer=10 (block-relay-only)
2022-06-28T07:06:21Z New outbound peer connected: version: 70016, blocks=742675, peer=11 (block-relay-only)
2022-06-28T07:07:45Z New outbound peer connected: version: 70016, blocks=742675, peer=12 (block-relay-only)
2022-06-28T07:14:00Z New outbound peer connected: version: 70016, blocks=742675, peer=16 (outbound-full-relay)

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-06-28 07:59:56 BST; 14min ago
    Process: 9112 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
   Main PID: 9350 (lnd)
[... 287 lines cut ...]
btc_default_sync_initial_done="1"
btc_default_sync_initialblockdownload="0"
btc_default_error_short=""
btc_default_sync_progress="0.9999944192436879"
btc_default_port="8333"
btc_default_blocks_headers="742675"
btc_default_version="v22.0.0"
btc_default_running="1"
btc_default_blocks_verified="742675"
btc_default_error_full=""
btc_default_peers="10"
btc_default_blocks_behind="0"
btc_default_sync_percentage="100.00"
btc_default_ready="1"
btc_default_mempool_transactions="3663"
btc_default_online="1"
btc_default_activated="1"
btc_default_synced="1"

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

stn021, 2022-06-28:
> Your lnd is locked. unlock with: `lncli unlock`

OK, I have manually unlocked the wallet now. I was not aware that I have to manually unlock the wallet

rootzoll, 2022-06-28:
@openoms is here something we should consider fixing/improving for v1.8.0 or was it more an edge case help session.

rootzoll, 2022-06-28:
OK so I will close this then. Please open a new issue for a future improvement if needed.

closed 2022-06-28 by rootzoll