raspiblitz/raspiblitz #3186: ip2tor with CLN

exitfiat, 2022-06-23:
Would be useful to have ip2tor/DuckDNS/Lets Encrypt working with Core Lightning. Was a great feature but Core is too fast and reliable to switch back to LND.

openoms, 2022-06-27:
Until IP2Tor with CLN is implemented you can just have an LND node wiith one private channel opened to your CLN one to be able to pay the invoices automatically.

Alternatively route around the whole Ip2Tor service by renting a VPS from http://host4coins.net or find more at https://bitcoin-vps.com.
On there you can do your own IP2Tor: 
https://github.com/openoms/bitcoin-tutorials/blob/master/tor2ip_tunnel.md
or tunnel through a VPN like ZeroTier (in SERVICES) or Tailscale (https://tailscale.com/download/linux).
Find some examples here about how to make some services accessible though the public IP of the VPS: https://github.com/openoms/bitcoin-tutorials/tree/master/nginx#nginx-scripts

exitfiat, 2022-06-27:
Thanks @openoms, this is very helpful

rootzoll, 2024-08-30:
PR to add Tailscale: #4712