raspiblitz/raspiblitz #3187: Reduce load by limiting LND to use 4 cores only

openoms, 2022-06-24:
New options in the lnd.conf to be tweaked accrodingly:
```
[workers]

; Maximum number of concurrent read pool workers. This number should be
; proportional to the number of peers. (default: 100)
; workers.read=200

; Maximum number of concurrent write pool workers. This number should be
; proportional to the number of CPUs on the host. (default: 8)
; workers.write=8

; Maximum number of concurrent sig pool workers. This number should be
; proportional to the number of CPUs on the host. (default: 8)
; workers.sig=4
```
https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf#L1139

Eg for the RPi4 with 4 cores:

```
[workers]
workers.write=4
workers.sig=4
```

Related: https://github.com/fusion44/blitz_api/issues/126

rootzoll, 2022-06-28:
PR merged. Set for final testing in v1.8.0rc3

rootzoll, 2022-07-27:
reports looked good over rc phase - closing for v1.8.0 release.

PR #3188 (2022-06-28): reduce lnd workers to number of cores if -lt 8
fix #3187
closed 2022-07-27 by rootzoll