raspiblitz/raspiblitz #3190: Universal auth for Core Lightning gRPC

bubelov, 2022-06-25:
CL gRPC looks pretty good since it eliminates the need to have any proxy deps. The only problem is auth, setting up a client requires:

- Tor hostname
- gRPC port
- Server cert
- Client cert
- Client private key

Obviously, Blitz knows all of that. Since there are no standards, what do you think about creating one and making it easily accessible from the menu? Having a single QR code with all of the required data would greatly simplify the auth flow

openoms, 2022-06-25:
Agree!
Related discussion: https://github.com/ZeusLN/zeus/issues/515

bubelov, 2022-06-26:
How about this:

```bash
#!/bin/bash

source /mnt/hdd/raspiblitz.conf

host=$(sudo cat /mnt/hdd/tor/clnGRPCport/hostname)
printf "host = %s\n" $host

port=$clnGRPCport
printf "port = %s\n" $port

server_pem=$(sudo cat -v /home/bitcoin/.lightning/bitcoin/server.pem | awk 'BEGIN{FS=OFS=";"}{gsub("\^M,","\\n",$1);gsub("\^M","",$1);print}')
printf "server_pem = %s\n" "$server_pem"

client_pem=$(sudo cat -v /home/bitcoin/.lightning/bitcoin/client.pem | awk '{printf "%s",$0}END{print}' | awk '{gsub("\^M","\\n",$0);print}')
printf "client_pem = %s\n\n" "$client_pem"

client_key_pem=$(sudo cat -v /home/bitcoin/.lightning/bitcoin/client-key.pem | awk '{printf "%s",$0}END{print}' | awk '{gsub("\^M","\\n",$0);print}')
printf "client_key_pem = %s\n\n" "$client_key_pem"

json=$(printf '{ "url": "%s:%s", "server_pem": "%s", "client_pem": "%s", "client_key_pem": "%s" }\n' "$host" "$port" "$server_pem" "$client_pem" "$client_key_pem" | jq)
printf "%s\n" "$json" | jq

qrencode -t utf8 "$json"
```

It produces the following JSON:

```json
{
  "url": "xxx.onion:4772",
  "server_pem": "-----BEGIN CERTIFICATE-----xxx-----END CERTIFICATE-----",
  "client_pem": "-----BEGIN CERTIFICATE-----xxx-----END CERTIFICATE-----",
  "client_key_pem": "-----BEGIN PRIVATE KEY-----xxx-----END PRIVATE KEY-----"
}
```

And then encodes is as a QR

If you have an Android phone, you can scan the QR with this app to see this login flow in action: https://github.com/bubelov/thunder/releases/tag/preview

I can refine this script and prepare a PR if there is no opposition to such an approach

bubelov, 2022-06-28:
Here is the cleaner version:

```bash
#!/bin/bash

source /mnt/hdd/raspiblitz.conf

host=$(sudo cat /mnt/hdd/tor/clnGRPCport/hostname)
port=$clnGRPCport
server_pem=$(sudo awk 'NF {sub(/\r/, ""); printf "%s\\n",$0;}' /home/bitcoin/.lightning/bitcoin/server.pem)
client_pem=$(sudo awk 'NF {sub(/\r/, ""); printf "%s\\n",$0;}' /home/bitcoin/.lightning/bitcoin/client.pem)
client_key_pem=$(sudo awk 'NF {sub(/\r/, ""); printf "%s\\n",$0;}' /home/bitcoin/.lightning/bitcoin/client-key.pem)
json=$(printf '{ "url": "%s:%s", "server_pem": "%s", "client_pem": "%s", "client_key_pem": "%s" }\n' "$host" "$port" "$server_pem" "$client_pem" "$client_key_pem")
```

rootzoll, 2024-04-23:
@openoms something to revisit for v1.11.1?

openoms, 2024-04-23:
Not sure if creating a new standard would be productive, but happy to implement if it is expected on a client side like in Zeus or else.