raspiblitz/raspiblitz #3193: [v1.8.0rc2] can't pay LN invoice using webUI

kravens, 2022-06-26:
The webUI does not properly recognize the time/time-out seconds for a Lightning invoice (from a BTC-payserver):
[image]

Using a different lightning wallet the same invoice could be paid, so the invoice format was correct.

How to reproduce:
1. Open raspiblitz webUI
2. Click Send (for lighting invoice)
3.  Paste invoice, click send
4.  Click Pay (found bug: Invoice expired: Jan 20, 1970, 5:04:13 AM)

openoms, 2022-06-26:
Thanks for the report! LND or CLN?

kravens, 2022-06-26:
With LND 15.0 beta

rootzoll, 2022-07-06:
When I try to pay the LND (1.8.0rc3) on my testnode a invoice the dialog just closes, nothing gets paid and I see in the logs:
```
Jul 06 23:24:01 raspberrypi python[770]: INFO:     *.*.*.*:36028 - "POST /lightning/send-payment?pay_req=lnbc12200n1p3vvz6jsp5204q4j3tlvsrkxtpal5sqh72u87amusq4gzznej5m95wf50uy9lqpp5fl83a554x9tjj2wedhru24k7fe8fhdsln35s67fh4l5725p4qa4sdzy2pshjmt9de6zqen0wgsrzv3jxqs8q6tcv4k8xgrpwss8xct5daeks6tn9ecxcctrv5hqxqzjccqpjrzjq00n7z3066l22s56t9jxrnncfjfzk2vp4ks6lzw0alxeena3dst2wzesasqqjscqqqqqqqlgqqqqqqgq9q9qyysgq8z3v06se7g0vfkn00pwp4xrzngau9rehpasd35pzqwdpsx45cnm4z7cg75n3rw32wwygwy7er94mmfqc42k0d5j4yc2kw0mfc36pyvsq4x9jtc&amount_msat=1220 HTTP/1.0" 400 Bad Request
```

@fusion44 what should I look for to debug deeper?

closed 2022-12-15 by rootzoll