raspiblitz/raspiblitz #3197: Sync of blockchain was fast but sync of lightning-wallet is taking days (2)

stn021, 2022-06-28:
Hello,

I think I have a better idea what was wrong. 

In the previous version of this issue I was informed that I have to manually unlock the lnd-wallet.

So far so good. **Except, that I did not set up lnd at all**. I specifically set up raspiblitz to use cln. 
So the lnd-wallet should not be a problem or even be mentioned.
That does sound inconsistent to me. Maybe there is something here to check after all.

And btw: after unlocking lnd the cln-wallet did sync in a reasonable time, even if this does not make sense.
But after then next reboot it started all over at 94%.

This happended with fresh sd-cards on at least two different nodes with different ssd and different sd-cards and different rpi-boards.

stn021, 2022-06-30:
I do not know. I flashed an sd-card, booted and followed the instructions.
Therefore what ever is in clconf is default.

> what is in your `clconf`?
I flashed the sd-card, booted and followed the menu. 
Therefore whatever is in `clconf` is the default and was not changed by me.
Any errors I mention appear after the initial setup, purely by the menu and before I have any opportunity to change anything at all.
I can post `clconf` , once you specifically tell me which file or program you mean.

> Is `rescan` present ?
Don't know what that is. I flashed the sd-card, booted and followed the setup-menu. Rescan is not mentioned anywhere.
I can see if rescan is present, once you specifically tell me where to look.

openoms, 2022-07-04:
I f you are restoring an old node the the config options are preserved. If you are storing from file or seed the `rescan` option is added to the CLN config file. You can find it with using the alias:
`clconf`
or in the SYSTEM menu.

cstenglein, 2023-06-07:
Seems resolved => closing

closed 2023-06-07 by cstenglein