raspiblitz/raspiblitz #3204: LiT Loop out Temporary Failure - not resolved by multiple restarts

redmac68, 2022-07-02:
Any advice would be really appreciated. 

I activated LND LiT service on my Raspiblitz 1.7.2. The activation seemed to work correctly (no errors?)
I had previously updated LND to v 0.15.0-beta. I had the LiT service activated previously, but  it was subsequently disabled.

I attempted two Loop Out's from local LiT web UI. Both eventually failed with error Temporary Failure ("The swap cannot continue due to an internal error. Manual intervention such as a restart is required."). I tried multiple re-starts of litd and full reboots of Raspiblitz. These did not resolve the issue.

The initial Loop Out seemed to proceed correctly until:

Jul 01 22:10:02 tatooine litd[3509934]: 2022-07-01 22:10:02.543 [INF] LOOP: f4b567 Swap script confirmed on chain
Jul 01 22:10:02 tatooine litd[3509934]: 2022-07-01 22:10:02.543 [INF] LOOP: f4b567 Htlc tx d4d1019278e40cb3e2c97f055f3834495bd529865da8b0478786a2ff7c74a629 at height 743208
Jul 01 22:10:02 tatooine litd[3509934]: 2022-07-01 22:10:02.544 [INF] LOOP: f4b567 Htlc value: 0.102 BTC
Jul 01 22:10:12 tatooine litd[3509934]: 2022-07-01 22:10:12.719 [ERR] LOOP: f4b567 Swap error: signing: rpc error: code = Unknown desc = selected sign method witness_v0 is not compatible with given pk script
Jul 01 22:10:12 tatooine litd[3509934]: 2022-07-01 22:10:12.720 [INF] LOOP: f4b567 Loop out swap state: FailTemporary
Jul 01 22:10:12 tatooine litd[3509934]: 2022-07-01 22:10:12.722 [ERR] LOOP: Execute error: signing: rpc error: code = Unknown desc = selected sign method witness_v0 is not compatible with given pk script

lit-loop monitor is now giving: [loop] rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: x509: certificate has expired or is not yet valid: current time 2022-07-02T10:25:42+01:00 is after 2022-03-27T13:46:43Z"

redmac68, 2022-07-03:
Disabled LiT and the re-enabled with LiT version updated to 0.7.0. Seems to be working now.

closed 2022-07-03 by redmac68

heikohaensel, 2022-11-19:
I had the same issue with my 1.7.2 installation. 
I only stopped the service litd, downloaded and replaced the binary files in /usr/local/bin with the ones of current LiT version 0.8.3-alpha.
Afterwards I started the service litd
The open loop out transaction was picked up by the daemon and it continued.