raspiblitz/raspiblitz #3205: Used the Update funciton on 1.7 and unable to launch raspiblitz

mantiev, 2022-07-02:
I get the below error

Starting SSH user interface ... (please wait)
/home/admin/_cache.sh: line 198: redis-cli: command not found
/home/admin/_cache.sh: line 198: redis-cli: command not found
setupPhase(done) state()
FAIL: /home/admin/raspiblitz.info does not exist or missing state.
Check logs & bootstrap.service for errors and report to devs.

It looks like it wiped out some needed info.

Appreciate any assistance.

Thanks.

mantiev, 2022-07-03:
I have looked at the errors and matched them with the correct file and everything looks correct.  I tried to upgrade to 1.7.2 and it didn't recognize my admin password.  At this point my node is completely down.

mantiev, 2022-07-04:
Thanks for the reminder.  That allowed me to login and setup the node.  I'm back up and running.  Thank you again.

closed 2022-07-04 by mantiev

souladventurer, 2022-10-13:
Hey @mantiev, I'm having the same issue and Im a bit of a dummy. My password seems to work but Im stuck in the loop of getting the same message every time I try and log in. How did you resolve?