raspiblitz/raspiblitz #3215: LIT Node Connect

mark0st83, 2022-07-08:
Hello,

I have problems with Lightning Terminal Node Connect. After copy pairing phrase and copy into the field on terminal.lightning.engineering there happens nothing. 
Looking into the logs shows me the connection tried to complete but there may be an issue.

`torsocks[1757173]: 2022-07-08 12:31:26.556 [WRN] GRPC: [core] grpc: Server.Serve failed to create ServerTransport:  connection error: desc = "transport: http2Server.HandleStreams failed to receive the preface from client: error decrypting payload: error reading from go-back-n: EOF" `

`WRN] GRPC: [core] grpc: Server.Serve failed to complete security handshake from "mailbox:Xxxxxxxxxxxxxxx@mailbox.terminal.lightning.today:443": error reading from go-back-n: receive timeout `

`torsocks[1757173]: 2022-07-08 12:36:17.627 [DBG] MBOX: Server: got failure on receive socket, re-trying: rpc error: code = Canceled desc = context canceled `

I have reinstalled it and also updated to 0.6.7-alpha. I didn’t know where to look for the problem.

mark0st83, 2022-07-12:
Yeah I’m already updated Tor to 0.4.7.8 one or two weeks ago. That’s because I had massive problems with peer-connections. But after I switched from Umbrel to raspiblitz I haven’t connected with lit yet.

openoms, 2022-07-20:
testing with the new version and getting the error:
[image]
need to trigger the creation of lit macaroons

guggero, 2022-07-21:
> @guggero could you please suggest how can we have the lit.macaroon created?

We did change a few things in LiT recently related to macaroons and LNC. So my answer would've been to update LiT as well (which you already did now).
Good to hear things are working now!

closed 2022-07-27 by rootzoll