raspiblitz/raspiblitz #3216: startup problems after prolonged shutdown

tanka1035, 2022-07-08:
Shutdown node for a couple weeks. Restarted and bitcoind, lnd, and electrum don't appear to be running. I've reflashed the sd and still having the same problem. Not sure where to start. My gut tells me there is a problem with the ssd or with a config. Please help. 

debug at https://l.termbin.com/yzm9r

NilByte, 2022-07-08:
Looks like db corruption on SSD from the log:

```
2022-07-08T16:21:32Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2022-07-08T16:21:32Z Fatal LevelDB error: Corruption: 1 missing files; e.g.: /mnt/hdd/bitcoin/chainstate/1245611.ldb
2022-07-08T16:21:32Z You can use -debug=leveldb to get more complete diagnostic messages
2022-07-08T16:21:32Z Fatal LevelDB error: Corruption: 1 missing files; e.g.: /mnt/hdd/bitcoin/chainstate/1245611.ldb
2022-07-08T16:21:32Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
```

Check this on how to reindex from command line:

https://github.com/rootzoll/raspiblitz/issues/3016

tanka1035, 2022-07-08:
About half way through reindexing now. Will follow up when it's finished. 

Thanks for the help.

tanka1035, 2022-07-11:
Here's the latest. I hope this doesn't mean I need to re-download the blockchain...

2022-07-11T00:14:17Z UpdateTip: new best=0000000000000000000c1e9e9b601972146aa48803500c2521a6f813df55d434 height=668162 version=0x20000000 log2_work=92.627252 tx=610812296 date='2021-01-29T08:30:51Z' progress=0.835357 cache=269.2MiB(1964147txo)
2022-07-11T00:14:17Z LevelDB read failure: Corruption: block checksum mismatch: /mnt/hdd/bitcoin/chainstate/618168.ldb
2022-07-11T00:14:17Z Fatal LevelDB error: Corruption: block checksum mismatch: /mnt/hdd/bitcoin/chainstate/618168.ldb
2022-07-11T00:14:17Z You can use -debug=leveldb to get more complete diagnostic messages
2022-07-11T00:14:18Z Error: Error reading from database, shutting down.
2022-07-11T00:14:18Z Error reading from database: Fatal LevelDB error: Corruption: block checksum mismatch: /mnt/hdd/bitcoin/chainstate/618168.ldb

closed 2022-07-12 by tanka1035