raspiblitz/raspiblitz #3217: BTCPay 1.6.0 Update

rootzoll, 2022-07-08:
https://twitter.com/BtcpayServer/status/1545444707169636353?s=20&t=zvQ7vNt2AhIcSYuwHpAlWw

openoms, 2022-07-09:
Updated ok from the menu in v1.7.2.

There is now a warning in the dashboard about needing the postgres backend for NBXplorer:

```
Wallet Balance
x.xxxx BTC

We would like to show you a chart of your balance. Please [migrate to the new NBXplorer backend](https://github.com/dgarage/NBXplorer/blob/master/docs/Postgres-Migration.md) for that data to become available.
```

Would like to have some help with autogeneratingt the postgres instance: #3055

openoms, 2022-07-09:
To use the latest script to update BTCpay on v1.7.2:
```
# download
wget -O /home/admin/config.scripts/bonus.btcpayserver.sh https://github.com/rootzoll/raspiblitz/blob/dev/home.admin/config.scripts/bonus.btcpayserver.sh

# update
/home/admin/config.scripts/bonus.btcpayserver.sh update
```

rootzoll, 2022-07-27:
Install looks good with new version - closing issue for v1.8.0

closed 2022-07-27 by rootzoll