raspiblitz/raspiblitz #3218: Add postgres backend for NBXplorer / BTCPayServer

openoms, 2022-07-09:
So far we have an older version of NBXplorer pinned: #3055 #3217 

Help would be appreciated to have postgres set up automatically. 

Keep in mind that postgres can potentially serve the lightning instances as well.

NicolasDorier, 2022-07-10:
just bump nbxplorer, and add those env variables: (replace with your own connection string if database is different)

```
      NBXPLORER_POSTGRES: User ID=postgres;Host=postgres;Port=5432;Application Name=nbxplorer;MaxPoolSize=20;Database=nbxplorer${NBITCOIN_NETWORK:-regtest}
      NBXPLORER_AUTOMIGRATE: 1
      NBXPLORER_NOMIGRATEEVTS: 1
```

Note that AUTOMIGRATE will just be ignored for new installs, so you can safely set that up for all.

In BTCPay, you need to link the database with the EXPLORERPOSTGRES.

```
BTCPAY_EXPLORERPOSTGRES: User ID=postgres;Host=postgres;Port=5432;Application Name=btcpayserver;MaxPoolSize=80;Database=nbxplorer${NBITCOIN_NETWORK:-regtest}
```

You know it's working if you can see btcpayserver wallet graph.

NicolasDorier, 2022-07-10:
version is `nicolasdorier/nbxplorer:2.3.28`

NicolasDorier, 2022-07-10:
Ping me if you have any issues. Though I didn't have any issues since I auto migrated people. Got a few perf issues on some big instances, but this has been fixed, so I doubt you'll have any issue.

openoms, 2022-07-11:
> version is `nicolasdorier/nbxplorer:2.3.28`

we are using https://github.com/dgarage/NBXplorer. That's the right repo, right?
https://github.com/dgarage/NBXplorer/releases/tag/v2.3.28

NicolasDorier, 2022-07-11:
yes, sorry I was linking the docker image

openoms, 2022-07-11:
The command line arguments for these (using them in the systemd service):
```
NBXPLORER_POSTGRES: User ID=postgres;Host=postgres;Port=5432;Application Name=nbxplorer;MaxPoolSize=20;Database=nbxplorer${NBITCOIN_NETWORK:-regtest}
NBXPLORER_AUTOMIGRATE: 1
NBXPLORER_NOMIGRATEEVTS: 1
```
```
--postgres User ID=postgres;Host=postgres;Port=5432;Application Name=nbxplorer;MaxPoolSize=20;Database=nbxplorermainnet
--automigrate 1
--nomigrateevts 1
```

NicolasDorier, 2022-07-11:
yes. Note systemd can also set environment variable.  (be careful there are spaces in the connection string)

openoms, 2022-07-11:
@NicolasDorier are you planning continue to PGP sign the BTCpay releases as well (had some signatures form Kukks before)?

NicolasDorier, 2022-07-11:
you mean for signing the release? I don't pay attention to it, git commit is signing automatically. Sometimes Kukks can make releases.

openoms, 2022-07-11:
> you mean for signing the release? I don't pay attention to it, git commit is signing automatically. Sometimes Kukks can make releases.

yes, we verify the tags - which is really just the signature of the last git commit. The issue is this being a random person (from a PR maybe) is that we cannot verify the updates.

openoms, 2022-07-11:
I used the setting for postgres in the /home/btcpay/.nbxplorer/Main/settings.config for NBXplorer:
```
postgres='User ID=postgres;Host=postgres;Port=5432;Application Name=nbxplorer;MaxPoolSize=20;Database=nbxplorermainnet'
automigrate=1
nomigrateevts=1
```
Getting the error:
```
info: NBXplorer.DatabaseSetup: Postgres services activated
dotnet[3263134]: Unhandled exception. System.ArgumentException: Format of the initialization s
tring does not conform to specification starting at index 84.    
```

Will need to initialize the postgres database. Do you have a script for that already?

NicolasDorier, 2022-07-11:
no I guess the problem come from the quote `'`

NicolasDorier, 2022-07-11:
@openoms if you use `settings.config`, remove quote.

openoms, 2022-07-11:
ok, works without quotes..
to fix 
```
Unhandled exception. System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (00000005, 0xFFFDFFFF): Name or service not known
```
needed to set the 
`Host=postgres;` to `Host=localhost;` 
since not dockerised.

but now expects a password:
```
Unhandled exception. Npgsql.NpgsqlException (0x80004005): No password has been provided but the backend requires one (in MD5)
```
It would be ideal to password protect the btcpay database.

will also need to:
- [ ] symlink the postgres data directory (`/var/lib/postgresql/13/main/`) to the disk

openoms, 2022-07-11:
What worked with a hardcoded password `raspiblitz` so far:

Install postgres:
```
sudo apt install -y postgresql
sudo systemctl enable postgresql
sudo systemctl start postgresql
```
Autogenerate the database:
```
sudo -u postgres psql -c "create database nbxplorermainnet;"
sudo -u postgres psql -c "create user btcpay with encrypted password 'raspiblitz';"
sudo -u postgres psql -c "alter user btcpay with encrypted password 'raspiblitz';"
sudo -u postgres psql -c "grant all privileges on database nbxplorermainnet to btcpay;"
```

NBXplorer config:
```
echo "\
network=mainnet
btc.rpc.user=raspibolt
btc.rpc.password=PASSWORD_B
postgres=User ID=btcpay;Host=localhost;Port=5432;Application Name=nbxplorer;MaxPoolSize=20;Database=nbxplorermainnet;Password='raspiblitz';
automigrate=1
nomigrateevts=1
" | sudo -u btcpay tee  /home/btcpay/.nbxplorer/Main/settings.config
```
```
₿ sudo journalctl -fu nbxplorer -n 100
...
info: NBXplorer.PostgresMigration: The migration completed in 0 minutes.
...
```

closed 2022-07-27 by rootzoll