raspiblitz/raspiblitz #3221: swapfile broken

dipunm, 2022-07-12:
Hey all,

Not sure when or how this happened, but the dphys-swapfile service keeps failing.
I figured it happened when I updated from v1.7 to v1.7.2

I looked at my partitions and /mnt/temp is VFAT
I went through the steps manually to create a swapfile (reading the bash scripts) and I think I found the issue:

```
admin# dd if=/dev/zero of=/mnt/temp/swapfile count=4096 bs=1MiB
dd: error writing '/mnt/temp/swapfile': File too large
```

The command does end up creating a file:
```
-rwxr-xr-x 1 bitcoin bitcoin 4294967295 Jul 12 03:00 swapfile
```

So maybe this is a red herring, but I think that is probably the root cause, I'm not sure whether I should change the format of the partition, or reduce the swapfile size?

dipunm, 2022-07-12:
This is my /etc/dphys-swapfile:
```
# /etc/dphys-swapfile - user settings for dphys-swapfile package
# author Neil Franklin, last modification 2010.05.05
# copyright ETH Zuerich Physics Departement
#   use under either modified/non-advertising BSD or GPL license

# this file is sourced with . so full normal sh syntax applies

# the default settings are added as commented out CONF_*=* lines

# where we want the swapfile to be, this is the default
CONF_SWAPFILE=/mnt/temp/swapfile

# set size to absolute value, leaving empty (default) then uses computed value
#   you most likely don't want this, unless you have an special disk situation
#CONF_SWAPSIZE=100

# set size to computed value, this times RAM size, dynamically adapts,
#   guarantees that there is enough swap without wasting disk space on excess
#CONF_SWAPFACTOR=2

# restrict size (computed and absolute!) to maximally this limit
#   can be set to empty for no limit, but beware of filled partitions!
#   this is/was a (outdated?) 32bit kernel limit (in MBytes), do not overrun it
#   but is also sensible on 64bit to prevent filling /var or even / partition
CONF_MAXSWAP=4096
```

Yet, when running `dphys-swapfile setup`, it says:
```
computing size, want /mnt/temp/swapfile=15744MByte, restricting to config limit: 4096MBytes, checking existing: deleting wrong size file (4294967295), generating swapfile ...
```
And then dies. Note that it wants a swapfile much larger than 4096MBytes?!

Not sure where that is coming from, but that might be the problem.

dipunm, 2022-07-12:
FYI, temporarily, I was able to get it all working by setting `CONF_MAXSWAP` to 4095 and by using `count=4095` in the `dd` command.

rootzoll, 2022-07-12:
@openoms is there something on the swap config we should adjust for the future based on this experience?

openoms, 2022-07-12:
Good catch! Let's reduce it. The swapfile on the disk is a backup only anyway since ZRAM has a higher priority.

openoms, 2022-07-19:
Note, that this is only an issue if using BTRFS (currently not offered and maintenance is experimental).
Still would have no problem with reducing the swapsize everywhere.

dipunm, 2022-07-19:
Was this line a mistake? https://github.com/rootzoll/raspiblitz/commit/264ee95ae4f9eab6628c79ad35d523becffacee8#diff-197b67bd151fcf88f239ccf6923eefbad6bff253305a3f4ec58d28380410b6d2R1465

rootzoll, 2022-07-27:
OK fixes so far will be inclused in v1.8.0 release - to be revisited with v1.8.1 - if no further reports it can be closed then,

closed 2022-12-15 by rootzoll