raspiblitz/raspiblitz #3224: JAM v0.0.8 needs NPM and node update

openoms, 2022-07-12:
```
##########################################################################
# OK --> the PGP signature of the checked out a958ba6 commit is correct
##########################################################################

npm ERR! code EBADENGINE
npm ERR! engine Unsupported engine
npm ERR! engine Not compatible with your version of node/npm: jam@0.0.8
npm ERR! notsup Not compatible with your version of node/npm: jam@0.0.8
npm ERR! notsup Required: {"node":">=16.11.0","npm":">=8.0.0"}
npm ERR! notsup Actual:   {"npm":"7.24.0","node":"v16.10.0"}
```

openoms, 2022-07-12:
We are 2 major versions behind with NodeJS:
[image]
Can we update to the latest or at least a v0.17.x @rootzoll @cstenglein ?

rootzoll, 2022-07-12:
The nodeJS v18 line seems to be the best to switch to and will be active until Oct next year:

https://nodejs.org/en/about/releases/
[image]

I will basically create a sd card build with v18 and see if any problems come up.

openoms, 2022-07-15:
Also npm can be installed as latest, no need to stick to v7 any more

openoms, 2022-07-15:
Settled on a more conservative nodeJS update which works with JAM also and has lower risk of breaking anything else.

rootzoll, 2022-07-27:
Looks good so far - closing issue for v1.8.0

closed 2022-07-27 by rootzoll