raspiblitz/raspiblitz #3230: Make installation and configuration of LNDg possible

md-weber, 2022-07-14:
I would love to have the possibility to install and configure LNDg right from the main menu. 

https://github.com/cryptosharks131/lndg

mark0st83, 2022-07-17:
I've installed lndg with 1.7.2 and a virtual setting on debian (proxmox). I've used @openoms [documentation ](https://github.com/openoms/lightning-node-management/blob/en/hardware/raspiblitz/lndg.md)for this.
I had to change a few things in the nginx.sh and a few commands from @openoms 

But it works

allyourbankarebelongtous, 2022-09-08:
I'm working on this now, see https://github.com/allyourbankarebelongtous/raspiblitz/tree/integrate_lndg. Still hasn't been tested yet, should get to that this weekend. After successful testing I'll submit a PR.

rootzoll, 2022-09-13:
PR #3313 is candidate to be merged in v1.8.1

rootzoll, 2022-12-09:
Merging #3313 into feature branch `3313-lndg` to test and finish last smal things.

rootzoll, 2022-12-09:
Make sure to still fix this: https://github.com/rootzoll/raspiblitz/pull/3313#discussion_r1024539631

rootzoll, 2022-12-09:
OK recovery test worked - merging `3313-lndg` branch. And to be final tested during RC phase.

closed 2022-12-15 by rootzoll