raspiblitz/raspiblitz #3245: AMD64 image Switch off sleep, suspend and hibernation, and temp monitor

moskusminer, 2022-07-26:
Hi! 
Ive just flashed raspiblitz amd64 image on to my mini-pc node project

Hardware: MSI Cubi-N 8GB RAM, (125GB SSD OS) (1TB Blockchain SSD) both internal disks

Everything booted perfectly, started to copy blockchain from my rpi4 over local network, after a while it was interrupted when the pc went into sleep mode.. 
is it possible to build this command in to the script from the start?
`sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target`

also my node is showing constant temp (not actual temp) in the info page, is there a way for me to change this in a conf file?

openoms, 2022-08-15:
The temperature options are discussed here: https://github.com/rootzoll/raspiblitz/issues/3283

rootzoll, 2022-09-29:
PR merged - closing issue.

closed 2022-09-29 by rootzoll