raspiblitz/raspiblitz #3246: LND-Backup process stopping mid-way during update

epther, 2022-07-27:
Trying to update Raspiblitz (currently running v1.7.0) to the latest version (v1.7.2).

When running the update wizard through the main menu (selecting the “Release” option), the CLI gets stuck on the LND-Backup step. The LND.RESCUE file does not appear to be created, it stops halfway.

Screen on Raspiblitz notes that LND is locked. Cancelling the process and logging back into Raspiblitz prompts for "Password C".

`*************************************
* JUST MAKING BACKUP TO OLD SD CARD
*************************************
please wait ..
# *** LND.RESCUE --> BACKUP
# Stopping lnd...
Job for lnd.service canceled.
# OK

tar: Removing leading `/' from member names
/mnt/hdd/lnd/
/mnt/hdd/lnd/v3_onion_private_key
/mnt/hdd/lnd/logs/
/mnt/hdd/lnd/logs/bitcoin/
/mnt/hdd/lnd/logs/bitcoin/mainnet/
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.12758.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.12756.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.12757.gz
/mnt/hdd/lnd/tls.key
/mnt/hdd/lnd/tls.cert
/mnt/hdd/lnd/data/
/mnt/hdd/lnd/data/watchtower/
/mnt/hdd/lnd/data/watchtower/v3_onion_private_key
/mnt/hdd/lnd/data/watchtower/bitcoin/
/mnt/hdd/lnd/data/watchtower/bitcoin/mainnet/
/mnt/hdd/lnd/data/watchtower/bitcoin/mainnet/watchtower.db
/mnt/hdd/lnd/data/chain/
/mnt/hdd/lnd/data/chain/bitcoin/
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/admin.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/invoice.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/router.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/chainnotifier.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/macaroons.db
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/readonly.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/charge-lnd.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/wallet.db
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/walletkit.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/signer.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/invoices.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
/mnt/hdd/lnd/data/graph/
/mnt/hdd/lnd/data/graph/mainnet/
/mnt/hdd/lnd/data/graph/mainnet/wtclient.db
/mnt/hdd/lnd/data/graph/mainnet/sphinxreplay.db
/mnt/hdd/lnd/data/graph/mainnet/channel.db`

rootzoll, 2022-07-27:
Backup of channel.db can take some time. If its a constant hold - use CTRL+C to stop the script. Might use command `raspiblitz` to return to menu. Reboot and see if node is still in a stable state. Then just normally shutdown and boot up again with an sd card containing the new updated image ... thats basically the update procedure without the wizard making a backup. Brst to use a second fresh sd card for the updfate and to keepthe old one as backup just in case.

epther, 2022-07-27:
Thanks rootzoll. I managed to fix this by using a different workaround. I followed @openoms '[Compact the LND database (channel.db)](https://www.lightningnode.info/technicals/lnddatabasecompaction)' tutorial. Turns out my channel.db file was 13GB, after I compacted it the file went down to ~300MB. 

I managed to save the LND-Backup file and update went smoothly thereafter.

woeisme, 2022-08-09:
I too have a massive Channel.db file and had to interrupt the backup script, perhaps a size check is in order and a warning to the user is needed?