raspiblitz/raspiblitz #3251: don't use sync-freelist=true in lnd.conf

openoms, 2022-07-28:
https://twitter.com/c_otto83/status/1552337039957413888

We can remove this option from the lnd.conf going forward as has a potential to cause problems.

Supposed to speed up the startup, but can cause high load in case of many channels.

To be clear I didn't experience force closes on v0.15. It is probably affecting nodes with many channels whuch shouldn't run on an RPi anyway.

TrezorHannes, 2022-07-29:
I've experienced both more FCs and an extremely fast growing channel.db with `sync-freelist=true`.
Ran both with true and off for 2 weeks, and can validate it was this single setting.
Since false, my channel.db grew by 100MB in 7 days, vs before 600MB-1GB per day, with 120 channels.

rootzoll, 2022-08-03:
@openoms do you think we should have this as default for new users or kind of force-change this with the next update?

openoms, 2022-08-03:
Yes, I think it can be removed from the default and if present should be changed to false with lnd.check.sh . I keep it off now and the node restarts are roughly the same.

rootzoll, 2022-12-13:
Final Testing: On release candidate check that it gets removes on update from v1.8 to v1.9

closed 2022-12-15 by rootzoll