raspiblitz/raspiblitz #3253: Just a suggestion....

DrBlackross, 2022-07-28:
for the https://github.com/rootzoll/raspiblitz/tree/dev/alternative.platforms#amd64-image

try installing the liquorix kernel, with ssd... for some reason the instructions suggest a usb drive (took forever to boot, then realized it was just how i was reading it), once i just wrote the image a 120gb ssd and expanded the system on my desktop..... BOOOM! boot up in 5 seconds. now with a liquorix kernel (on amd), i see the grub for 3 seconds then instant gnome login manager, crazy fast.

DrBlackross, 2022-07-28:
and yes, once i migrate my NES rpi node to this wicked AMD rack mount job, i will retire my NES node

DrBlackross, 2022-07-28:
Oh yeah, and turn off suspend lol
$ sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target

BitromNO, 2022-07-29:
its fast! no doubt! i use a MSI Cubi-N mini pc, with internal disk for os and blockchain, and im so happy with it. no more cabels sticking out! 
just patch it to v.1.8 in the patch menu. the version came out yesterday :)

BitromNO, 2022-07-29:
> Oh yeah, and turn off suspend lol $ sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target

i beleve they are going to make this ready in v1.8.1, ive alreasy made an issue: 
https://github.com/rootzoll/raspiblitz/issues/3245

DrBlackross, 2022-07-29:
yea was kinda fun, had it all set up in the rack mount case, dragged it down to the basement, mounted it into the 2nd (wall house) rack, ran power and net.... fired it up.

came upstairs, its pinging out LOL

go back down look over across the basement and the power led is flashing like a jack russell :0

wchancao, 2022-07-29:
Did you connect the nes pi data ssd into the pc? Or did you sync a new node with a new ssd?

DrBlackross, 2022-07-29:
i had 5.25 hdd 2tb to use, its a server hd, sooooo its mounted in a server case LOL. the main boot drive is a 120gb ssd kingston (that has raspiblitz on it with filesystem expanded)

DrBlackross, 2022-07-29:
your would think after 30+ years of bulletin board systems..... i would be used to this. but i still post like its an irc chat

DrBlackross, 2022-07-29:
> Did you connect the nes pi data ssd into the pc? Or did you sync a new node with a new data ssd?

BUT, i did use my .bitcoin folder on my desktop (linux) to move the blockchain over to the Nespi before in the past... it does speed it up, the RPI is a bit slow when it comes to tor, with the AMD in the basement im already at 25% sync'd after a day, with the Nespi's RPI4 it took a week for me to get to that percent 

(i don't have a mobo for my threadripper or i'd thrown it on that LOL

DrBlackross, 2022-07-29:
i do miss the LTC lightning node

DrBlackross, 2022-08-11:
hmmmm....
after removing blitzapi (webgui)
this few by the screen.

dialogcancel(0) (19)
Blitz API + webUI settings changed ..
# UNINSTALL Web API
Failed to stop blitzapi.service: Unit blitzapi.service not loaded.
Failed to disable unit: Unit file blitzapi.service does not exist.
rm: cannot remove '/etc/systemd/system/blitzapi.service': No such file or directory
userdel: user 'blitzapi' does not exist

cstenglein, 2023-06-07:
@DrBlackross is this still relevant? Closing for now, if it's still relevant, please re-open

closed 2023-06-07 by cstenglein

DrBlackross, 2023-07-19:
Honestly i forgot about it, been out of country off and on too much, but it worked fine while i was away. i did upgrade to 1.9 through the menu, still working fine but cant remember how to move fund from chain balance to channel balance lol

btw, did you drop the ltc/lnd node build?

DrBlackross, 2023-07-20:
BTW, amd64 1.10.0rc1 works (good so far on fresh reinstall) BUT....
 
Err:6 http://archive.raspberrypi.org/debian bullseye InRelease                   
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E

you need to remove out the raspi apt sources /\
i was trying to push liquorix kernel on the "new" setup after it rebooted.

also during new install with old blockchain hdd (and your supplied kernel, i re-imaged the previous main drive with the raspiblitz-amd64-debian-lean.img 1.10.0.rc1) i saw these fly by a few times..

/home/admin/config.scripts/bonus.bos.sh: line 213: cd: /home/bos/balanceofsatoshis: Permission denied
/home/admin/config.scripts/bonus.thunderhub.sh: line 103: cd: /home/thunderhub/thunderhub: Permission denied
/home/admin/config.scripts/bonus.mempool.sh: line 138: cd: /home/mempool: Permission denied
fatal: could not create work tree dir 'mempool': Permission denied
/home/admin/config.scripts/bonus.mempool.sh: line 140: cd: mempool: Permission denied
fatal: not a git repository (or any of the parent directories): .git
sudo: unable to execute /home/admin/config.scripts/blitz.git-verify.sh: Permission denied

(was kinda weird, ill check it after it mostly settles)

but so far, i like it! (well on the rpi4 it looked cool)

DrBlackross, 2023-07-20:
had to re-chown everything in /home/admin to current admin, that fixed the permission denied

sudo chown admin:admin -R /home/admin

then went into menu and updated bos, etc....

DrBlackross, 2023-07-21:
Heads up, fresh install by imaging with balenaEtcher-1.5.100-x64 of raspiblitz-amd64-image-2023-05-23-e776671.zip (and extra steps to *.img) to the main hdd works great, once the node is fully sync'd i'll try importing the lnd-rescue.

DrBlackross, 2025-11-16:
and then using the migration file

[raspiblitz.log](https://github.com/user-attachments/files/23569681/raspiblitz.log)

DrBlackross, 2025-11-21:
took a few to get it to take the migration file, i think it worked, but i wont know till the node is done syncing