raspiblitz/raspiblitz #3254: v1.8.0: Setup of new device is stuck when aborting COPY blockchain

freerko, 2022-07-28:
Hi,

just downloaded the fresh 1.8.0 SD card image and used it with a brand new SSD in a Raspberry Pi 4. I wanted to copy the blockchain from my laptop via USB to the SSD, so I ran the initial RaspiBlitz setup up until the "Blockchain Sync/Validation" screen and choosed COPY. I was expecting that RaspiBlitz just creates the initial directory structure so that I can see where and with which ownership to copy the blockchain to. But after selecting Linux as my OS I got a screen telling me to rsync the data, but since I wanted to copy it directly via USB instead of via the network I canceled with CTRL+C as advised in the screen. Then some Quickcheck told me that there was an invalid transfer and if I want to "reset/delete data". Didn't really understand what that implies and so I choosed Yes. After the automatic reboot I just get "Blockchain Progress : 5 restarts" and there is no /mnt/hdd/bitcoin directory. Another reboot doesn't change anything.

It would be great if the setup wouldn't be stuck after aborting the COPY blockchain screen. Would it be possible to just run the initial setup wizard again in this case?

BitromNO, 2022-07-29:
i had the same problem. just flash a new ssd and try again.
local copy is fast compared to p2p. around 3h.

rootzoll, 2022-08-21:
TODO for v1.8.1 give notice on failed quickcheck (when zero data was transfered) how to proceed.

PR #3526 (2022-12-15): #3254 add info
fixes #3254
closed 2022-12-15 by rootzoll