raspiblitz/raspiblitz #3255: 1.8.0: no SSE messages if blocks are syncing

cstenglein, 2022-07-29:
Usually, if the blockchain is syncing, the WebUI shows a Syncscreen.

My Raspiblitz was at 100%, but due to the load, had to catch up with the blockchain.

If the blockchain is syncing, no messages arrived over SSE, so there are no updates (not even hardware info etc.)

[image]

rootzoll, 2022-08-03:
The state in the redis is changed on sync (thats why the SSH & LCD is changing) but I thin kthe API does not reflect monitor this - @fusion44 is this correct? Do we wanna optimize this vor v1.8.1?

rootzoll, 2022-12-19:
To recheck during ReleaseCandidate phase of v1.9 (putting on final testing) - also try to fund a way to reproduce the szenario.

cstenglein, 2023-06-11:
@fusion44 Issue still appears currently.

I have CLN syncing currently, so it shouldn't let me into the webUI, but it did.

Weird thing is, I only get `btc_info` and `system_startup_info` messages, not the others. 

[image]

rootzoll, 2023-08-11:
@cstenglein & @fusion44 what was the status of this issue - is it still relevant?

cstenglein, 2023-08-12:
Not sure since it's hard to reproduce.

We can close this and re-open if needed if you like.

closed 2023-08-13 by rootzoll