raspiblitz/raspiblitz #3256: Migration to new hardware

BitromNO, 2022-07-29:
My static backup settings to nextcloud does not work after migration, dont know if this is a bug?
and also is is better to flash a new sd card rather than patching from v1.8rc4 to v1.8?
ive patched two nodes today.. :)

BitromNO, 2022-07-29:
one of the nodes is installed with amd64 v1.7.2 image. 
do i have to build a amd64 v1.8 image myself to do a propper clean update?

rootzoll, 2022-08-21:
> do i have to build a amd64 v1.8 image myself to do a propper clean update?

@openoms whats the recommended update procedure for amd64 images?

openoms, 2022-08-21:
Working on the automatic building on the new image, will worth to wait for that: https://github.com/openoms/raspiblitz/actions/workflows/amd64-image-build.yml

Until then dirty update it is and reinstalling services to refresh.

closed 2023-06-07 by cstenglein