raspiblitz/raspiblitz #3258: Migrated from an umbrel 0.5.1 testnode to 1.8, and found the following 2 hic-ups

TrezorHannes, 2022-07-29:
Not sure if it's related to the umbrel move, or a general 1.8 issue

## 1) no output logging into /mnt/hdd/bitcoin/debug.log 
```
admin@192.168.86.245:~ ₿ sudo systemctl status bitcoind.service 
* bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-07-29 19:22:29 BST; 5min ago
    Process: 2032 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)q
```
I've restarted bitcoind and the node several times, but can't get bitcoin to write the log
```
admin@192.168.86.245:~ ₿ sudo tail -n 30 -f /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: no files remaining
admin@192.168.86.245:~ ₿ ls -la /mnt/hdd/bitcoin/  
total 14844
drwxrwxr-x  5 bitcoin bitcoin    4096 Jul 29 19:22 .
drwxr-xr-x 13 root    root       4096 Jul 29 19:18 ..
-rw-rw-r--  1 bitcoin bitcoin       0 Jan 19  2021 .gitkeep
-rw-------  1 bitcoin bitcoin  224355 May 26  2021 banlist.dat
-rw-------  1 bitcoin bitcoin      34 May 26 22:21 banlist.json
-rw-r--r--  1 bitcoin bitcoin     745 Jul 29 13:25 bitcoin.conf
drwx------  3 bitcoin bitcoin  184320 Jul 29 17:07 blocks
drwx------  2 bitcoin bitcoin  114688 Jul 29 19:29 chainstate
-rw-------  1 bitcoin bitcoin       0 Feb 25  2021 db.log
-rw-------  1 bitcoin bitcoin  247985 Jul 29 19:18 fee_estimates.dat
drwx------  4 bitcoin bitcoin    4096 Feb 25  2021 indexes
-rw-------  1 bitcoin bitcoin 8754466 Jul 29 19:18 mempool.dat
-rw-------  1 bitcoin bitcoin      99 Jul 29 19:22 onion_v3_private_key
-rw-------  1 bitcoin bitcoin 4137648 Jul 29 19:18 peers.dat
-rw-r--r--  1 bitcoin bitcoin       4 Jul 29 19:20 settings.json
-rw-------  1 bitcoin bitcoin 1490944 Feb 24 11:55 wallet.old.bak
```

## 2) auto-unlock shows cleartext in the menu
[image]

then I get prompted to enter PASSWORD C, but it's in cleartext
[image]

and I'm uncertain why a node-reboot is necessary after activating auto-unlock. Shouldn't it be sufficient to `sudo systemctl restart lnd.service`?

TrezorHannes, 2022-07-29:
It's 3 hic-ups:

## Auto-unlock doesn't work
Apparently the cleartext password entry and reboot didn't work out properly. Will dig in further but here's a first screenshot after the reboot

[image]

rootzoll, 2022-11-14:
Thanks for the report. The auto-unlock is basically storing your wallet password c in cleartext on the HDD - thats the security compromise. So displaying it in cleartext in the SSH menu was a design choice - its quicker to spot a typo.

Autounlock funxction is a task on our to fix list for next release - see #2356

Will keep open this issue until we have the first release candidates of next version to test if I can reproduce the "no output logging into /mnt/hdd/bitcoin/debug.log" on migration test umbrel -> raspiblitz

thx for the report.