raspiblitz/raspiblitz #3259: Link to Mempool didn't work anymore after updated from v1.7.1 to v1.8.0

rene78, 2022-07-29:
Steps:

1. Updated RaspiBlitz from 1.7.1 to v1.8.0
2. Tried to open mempool.space but realized that url has changed to a different IP address
3. When using the new url the request timed out
4. Then I went to the new webinterface and uninstalled the mempool.space app
5. Tried to reinstall within the webinterface but installation failed
6. Installation via the SSH menu suceeded then and everything works as before

Hardware: RaspiBlitz with metal case and 4GB RAM

rootzoll, 2022-08-03:
I think we need better error reporting on WebUI installs to see what goes wrong there. At least mempool app install to be retested - see if can be reproduced.

rootzoll, 2022-12-12:
better error reporting will be done in #3318 - closing this issue.

closed 2022-12-12 by rootzoll