raspiblitz/raspiblitz #3260: WebUI: Password is wrong

net47, 2022-07-30:
Hello,

refering to https://github.com/rootzoll/raspiblitz/issues/3145, I currently have the exact same problem: if I go to the WebUI and enter my password A, I get the error `An error occured: Password is wrong`

The password is definitely the correct one. Journal output: 

`Jul 30 10:54:42 raspberrypi python[822]: INFO:     127.0.0.1:47570 - "POST /system/login HTTP/1.0" 401 Unauthorized`

```
₿ /home/admin/config.scripts/blitz.passwords.sh check a "XXXXXXXXX"
# salt file exists
# hashedPasswordSalt(XXXXXXXXXX)
# first try within last minute - respond fast
correct=1
```

cstenglein, 2022-07-30:
Do you have Special characters in your password A?

net47, 2022-07-30:
> Do you have Special characters in your password A?

Yes, a minus and a point. I now changed it to only upper/lowercase+digits and login works now. Thanks for the hint!

thecharacter, 2022-08-01:
Hi Im having the same issue. No special characters in my password. Also command /home/admin/config.scripts/blitz.passwords.sh check a "XXXXXXXXX" 
returned correct=1

Tried Brave and Chrome browsers.

rootzoll, 2022-08-03:
@cstenglein will we update the password policy in the WebUI to also math the old SSH ruleset of including "-._" or is this an API thing?

cstenglein, 2022-08-03:
It's a webUI thing:
https://github.com/cstenglein/raspiblitz-web/blob/master/src/components/Settings/ChangePwModal/ChangePwModal.tsx#L97

Currently we don't allow any special characters, but I can change that easily

So it should be `a-zA-Z0-9` as well as `-`, `.`, `_` ?

rootzoll, 2022-08-03:
@cstenglein this is the line in the SSH script setting the password:
https://github.com/rootzoll/raspiblitz/blob/b20293a0c90f531477774c9a810b5491c81d0166/home.admin/config.scripts/blitz.passwords.sh#L218

So as special chars its: `-` & `.` that should be also allowed.

dnlggr, 2022-08-05:
> @cstenglein will we update the password policy in the WebUI to also math the old SSH ruleset of including "-._" or is this an API thing?

I do think this is an API thing too. 

It looks like the [API uses the same alphanum-only regex](https://github.com/fusion44/blitz_api/blob/main/app/repositories/system.py#L61) as the WebUI when checking a password upon login.

I'll open PRs for the WebUI and the API so that we can see if this fixes it.

tlindi, 2022-11-30:
Installed with script parameter "-b dev" onto Debian 11 x86_64 and i still encounter this same - admin (A)  password returns "An error occured: Password is wrong". But password is working in ssh connection.

any ideas?

cstenglein, 2022-12-17:
> Do you think we have it covered?

No.

The UI doesn't differentiate between 401 & 500 currently. https://github.com/cstenglein/raspiblitz-web/issues/403

So it shows "Password is wrong" even though the API might not work at all. Most likely the issue @tlindi has.

The other issue would be the API itself, which performs a check on the password as well. A user was trying to fix it here: https://github.com/fusion44/blitz_api/pull/141 but it was closed (most likely bc the account was deleted).

In my opinion, though, the password shouldn't be checked by the API but only the raspiblitz itself, so the check in the API should be removed.

cstenglein, 2022-12-17:
Another good issue to implement would be https://github.com/fusion44/blitz_api/issues/145 so there is a low cost check if the API is working at all.

rootzoll, 2022-12-19:
> The UI doesn't differentiate between 401 & 500 currently. https://github.com/cstenglein/raspiblitz-web/issues/403

I think its good to give back the HTTP error as a basic info: Timeout, 404, 500, ...

rootzoll, 2022-12-19:
@fusion44 OK PR is available https://github.com/fusion44/blitz_api/pull/181 - please review & merge

thecharacter, 2023-01-28:
I had to setup my v1.8 pi from scratch and notice that if I set the passwords with the web UI it works. Previously I was setting the passwords in the ssh terminal. Maybe the web ui and ssh terminal read characters differently?

rootzoll, 2023-01-28:
the v1.8 has some password check differences with the ssh - you can try the v1.9.0rc1 sd card image that should have that fixed: https://github.com/rootzoll/raspiblitz/tree/v1.9#downloading-the-software

closed 2023-05-08 by rootzoll