raspiblitz/raspiblitz #3261: Update instructions reference LCD which may not exist

orangesurf, 2022-07-30:
I am running my Raspiblitz without an LCD, the update instructions direct me to follow the LCD instructions. 

https://github.com/rootzoll/raspiblitz/blob/b20293a0c90f531477774c9a810b5491c81d0166/home.admin/99updateMenu.sh#L83

orangesurf, 2022-07-30:
LCD instructions unknown but LCD is not required as user can ssh in with the normal admin:192... and the default password of raspiblitz.

openoms, 2022-07-30:
That's right. Also with the new WebUI you can just type the raspiblitz IPaddress to the browser.
Can also try opening:
```
raspberrypi.local
```
in the browser.
but that won't always work depending on the router.

orangesurf, 2022-07-30:
Great. Love the WebUI look. Would it be useful to change the "Follow the instructions on the LCD" instruction to "ssh in with admin@ and password:raspiblitz "?

PR #4619 (2024-06-10): #3261 change text
fixes #3261
closed 2024-06-10 by rootzoll