raspiblitz/raspiblitz #3262: Webinterface doesn't receive data from the API

freerko, 2022-07-30:
Hi,

I've set up a new RaspiBlitz v1.8.0 on a Raspi4 with 8GB of RAM. I am able to login to the new web interface, but only see static stuff and spinning wheels:
[image]

Sometimes its just a spinning wheel (and after 10 minutes or so (!) it finally turns into the screenshot above):
[image]

In the settings, there is no information on RaspiBlitz or Blitz-API versions but the Reboot button works, so there has to be some communication to the API (or does that use a different connection?)
[image]

After a reboot its still the same. 

The Blitz_API seems to be running:
> root@raspberrypi:~# ps aux | grep api | grep -v grep
> blitzapi     783  2.2  0.7 929144 29628 ?        Ssl  13:03   0:46 /usr/bin/python -m uvicorn app.main:app --port 11111 --host=0.0.0.0 --root-path /api

But with this via SSH on the RaspiBlitz:

> curl --header "Content-Type: application/json" \
>  --request POST \
>  --data '{"password":"mypassword"}' \
>  http://127.0.0.1:11111/system/login

I get `{"detail":"Not Found"}`, with `curl -N http://127.0.0.1:11111/v1/bitcoin/getblockchaininfo` the same Not Found

I can't find any logfile for the blitz_api. How can I debug this further?

3773, 2022-07-31:
I have the same problem. I have already done the patch update.

rootzoll, 2022-08-03:
@cstenglein can you help debug this?

cstenglein, 2022-08-05:
I think the path should be `http://127.0.0.1:11111/v1/bitcoin/getblockchaininfo`

Otherwise I can look into It more next weekend.

cstenglein, 2022-08-19:
@freerko Maybe try to check what happened with the blitzAPI:

`sudo journalctl | grep 'BlitzBackendAPI' -A 100`

In the RaspiBlitz Telegram someone had a python issue which resulted in the api not installing a dependency correctly.

freerko, 2022-08-19:
Your command didn't return anything as the logs where full of other stuff from the last three weeks. 
Weird thing is, after a reboot now the API and Webinterface run without issues for the first time. I didn't change anything and already did several reboots after I've set it up three weeks ago, but now after one other reboot it just works 🤷‍♂️

cstenglein, 2022-08-19:
@rootzoll should we investigate that or leave it closed?

rootzoll, 2022-08-21:
I

rootzoll, 2022-08-21:
I reopned so that we can focus for v1.8.1 to improve on this again.

ChuckNorrison, 2022-10-06:
Same behaviour for me

`sudo ./blitz.web.api.sh update-code` results with some errors

> Building wheels for collected packages: psutil
>   Building wheel for psutil (setup.py) ... done
>   Created wheel for psutil: filename=psutil-5.9.2-cp39-cp39-linux_aarch64.whl size=284003 sha256=cf93688db71a43242f64f178c89598d06ab8df8aa62a00c5d329b26861f746f8
>   Stored in directory: /root/.cache/pip/wheels/90/b2/61/285f7409a9dd2ffa1aa0684602a619f61b4c7b41c7ba7ba7e3
> Successfully built psutil
> Installing collected packages: wrapt, urllib3, typing-extensions, tenacity, sniffio, setuptools, pyzmq, python-json-logger, psutil, protobuf, idna, h11, grpcio, frozenlist, charset-normalizer, certifi, attrs, yarl, uvicorn, requests, pydantic, grpcio-tools, googleapis-common-protos, deprecated, aioredis, redis, fastapi, fastapi-plugins
>   Attempting uninstall: wrapt
>     Found existing installation: wrapt 1.12.1
> ERROR: Cannot uninstall 'wrapt'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
> --- Logging error ---
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 177, in emit
>     self.console.print(renderable, overflow="ignore", crop=False, style=style)
>   File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1752, in print
>     extend(render(renderable, render_options))
>   File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1390, in render
>     for render_output in iter_render:
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 134, in __rich_console__
>     for line in lines:
>   File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/segment.py", line 245, in split_lines
>     for segment in segments:
>   File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1368, in render
>     renderable = rich_cast(renderable)
>   File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/protocol.py", line 36, in rich_cast
>     renderable = cast_method()
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 130, in __rich__
>     pip_cmd = get_best_invocation_for_this_pip()
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/entrypoints.py", line 58, in get_best_invocation_for_this_pip
>     if found_executable and os.path.samefile(
>   File "/usr/lib/python3.9/genericpath.py", line 101, in samefile
>     s2 = os.stat(f2)
> FileNotFoundError: [Errno 2] No such file or directory: '/usr/bin/pip3.9'
> Call stack:
>   File "/usr/local/bin/pip", line 8, in <module>
>     sys.exit(main())
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 70, in main
>     return command.main(cmd_args)
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 101, in main
>     return self._main(args)
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 223, in _main
>     self.handle_pip_version_check(options)
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/req_command.py", line 148, in handle_pip_version_check
>     pip_self_version_check(session, options)
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 237, in pip_self_version_check
>     logger.info("[present-rich] %s", upgrade_prompt)
>   File "/usr/lib/python3.9/logging/__init__.py", line 1442, in info
>     self._log(INFO, msg, args, **kwargs)
>   File "/usr/lib/python3.9/logging/__init__.py", line 1585, in _log
>     self.handle(record)
>   File "/usr/lib/python3.9/logging/__init__.py", line 1595, in handle
>     self.callHandlers(record)
>   File "/usr/lib/python3.9/logging/__init__.py", line 1657, in callHandlers
>     hdlr.handle(record)
>   File "/usr/lib/python3.9/logging/__init__.py", line 948, in handle
>     self.emit(record)
>   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 179, in emit
>     self.handleError(record)
> Message: '[present-rich] %s'
> Arguments: (UpgradePrompt(old='22.1.2', new='22.2.2'),)
> \# BRANCH ---> main
> \# old commit -> a4e9f7bf23e4b30e67aef3fb3c0208c11a2f1cd3
> \# new commit -> bd19c41fd55648efd4440fcc93c7d6928ef35f4b
> \# blitzapi updates and restarted

rootzoll, 2022-10-06:
> raspberrypi python[1712724]: WARNING:root:Key 'tor_web_addr' not found in Redis DB.

@fusion44 can this be a reason of the API falling into an excpetion and not delivering data

fusion44, 2022-10-08:
Hmm no, that's not it. The code just returns an empty string if a key is not found:

```python
async def redis_get(key: str) -> str:
    v = await redis_plugin.redis.get(key)

    if not v:
        logging.warning(f"Key '{key}' not found in Redis DB.")
        return ""

    return v.decode("utf-8")
```

From @ChuckNorrison's the logs above:
```
FileNotFoundError: [Errno 2] No such file or directory: '/usr/bin/pip3.9'
```
Maybe a system level Python problem?

ChuckNorrison, 2022-10-09:
python works for me, but the file pip3.9 does not exist on my system

> admin@192.168.8.160:~ ₿ python --version
> Python 3.9.2

ChuckNorrison, 2022-10-09:
Another error found in the blitzapi journal with the same reason

> Oct 09 23:26:20 raspberrypi systemd[1]: Started BlitzBackendAPI.
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:     Started server process [2134895]
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:uvicorn.error:Started server process [2134895]
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:     Waiting for application startup.
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:uvicorn.error:Waiting for application startup.
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:     Application startup complete.
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:uvicorn.error:Application startup complete.
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:     Uvicorn running on http://0.0.0.0:11111 (Press CTRL+C to quit)
> Oct 09 23:26:27 raspberrypi python[2134895]: INFO:uvicorn.error:Uvicorn running on http://0.0.0.0:11111 (Press CTRL+C to quit)
> Oct 09 23:26:27 raspberrypi python[2134895]: ERROR:asyncio:Task exception was never retrieved
> Oct 09 23:26:27 raspberrypi python[2134895]: future: <Task finished name='Task-7' coro=<_handle_gather_hardware_info() done, defined at ./app/repositories/system.py:170> exception=AttributeError("'str' object has no attribute 'decode'")>
> Oct 09 23:26:27 raspberrypi python[2134895]: Traceback (most recent call last):
> Oct 09 23:26:27 raspberrypi python[2134895]:   File "./app/repositories/system.py", line 173, in _handle_gather_hardware_info
> Oct 09 23:26:27 raspberrypi python[2134895]:     info = await get_hardware_info()
> Oct 09 23:26:27 raspberrypi python[2134895]:   File "./app/repositories/system.py", line 117, in get_hardware_info
> Oct 09 23:26:27 raspberrypi python[2134895]:     return await get_hardware_info_impl()
> Oct 09 23:26:27 raspberrypi python[2134895]:   File "./app/repositories/hardware_impl/raspiblitz.py", line 14, in get_hardware_info_impl
> Oct 09 23:26:27 raspberrypi python[2134895]:     loads = (await redis_get("system_cpu_load")).split(",")
> Oct 09 23:26:27 raspberrypi python[2134895]:   File "./app/utils.py", line 64, in redis_get
> Oct 09 23:26:27 raspberrypi python[2134895]:     return v.decode("utf-8")
> Oct 09 23:26:27 raspberrypi python[2134895]: AttributeError: 'str' object has no attribute 'decode'
> Oct 09 23:26:36 raspberrypi python[2134895]: INFO:     127.0.0.1:43368 - "GET /lightning/list-all-tx?reversed=true HTTP/1.0" 200 OK

ChuckNorrison, 2022-10-10:
Reflashed my sdcard with v1.8.0 and my WebUI works. Just one box with loading animation left

[image]

MarcelRapold, 2022-10-12:
I have the same error after migration to a proxmox vm, any ideas on how to fix this?

thanks a lot guys! @cercatrova21 

[image]

muraschal, 2022-10-13:
Update:

After a RaspiBlitz (Version 1.8.0b with lnd 0.15.2) Cleaninstall on Ubuntu, the Error is still persistent.
I'cant be the only one having this issues, can i manually fix this with a workaround?

thanks

cstenglein, 2022-10-14:
@ChuckNorrison the issue with the one loading box is in the webUI, I created an issue for that: https://github.com/cstenglein/raspiblitz-web/issues/493

muraschal, 2022-10-14:
[image]
after debugging session with @cstenglein we did not find a solution...

benjamin-wilson, 2022-10-14:
Clean Proxmox 1.8 CLN install:

```
Oct 14 08:04:03 debian python[3531]: ERROR:uvicorn.error:Exception in ASGI application
Oct 14 08:04:03 debian python[3531]: Traceback (most recent call last):
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/uvicorn/protocols/http/h11_impl.py", line 404, in run_asgi
Oct 14 08:04:03 debian python[3531]:     result = await app(  # type: ignore[func-returns-value]
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/uvicorn/middleware/proxy_headers.py", line 78, in __call__
Oct 14 08:04:03 debian python[3531]:     return await self.app(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/fastapi/applications.py", line 269, in __call__
Oct 14 08:04:03 debian python[3531]:     await super().__call__(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/applications.py", line 124, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.middleware_stack(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/middleware/errors.py", line 184, in __call__
Oct 14 08:04:03 debian python[3531]:     raise exc
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/middleware/errors.py", line 162, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, _send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/middleware/cors.py", line 84, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/exceptions.py", line 93, in __call__
Oct 14 08:04:03 debian python[3531]:     raise exc
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/exceptions.py", line 82, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, sender)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/fastapi/middleware/asyncexitstack.py", line 21, in __call__
Oct 14 08:04:03 debian python[3531]:     raise e
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/fastapi/middleware/asyncexitstack.py", line 18, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/routing.py", line 670, in __call__
Oct 14 08:04:03 debian python[3531]:     await route.handle(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/routing.py", line 418, in handle
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/fastapi/applications.py", line 269, in __call__
Oct 14 08:04:03 debian python[3531]:     await super().__call__(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/applications.py", line 124, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.middleware_stack(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/middleware/errors.py", line 184, in __call__
Oct 14 08:04:03 debian python[3531]:     raise exc
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/middleware/errors.py", line 162, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, _send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/exceptions.py", line 93, in __call__
Oct 14 08:04:03 debian python[3531]:     raise exc
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/exceptions.py", line 82, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, sender)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/fastapi/middleware/asyncexitstack.py", line 21, in __call__
Oct 14 08:04:03 debian python[3531]:     raise e
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/fastapi/middleware/asyncexitstack.py", line 18, in __call__
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/routing.py", line 670, in __call__
Oct 14 08:04:03 debian python[3531]:     await route.handle(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/routing.py", line 266, in handle
Oct 14 08:04:03 debian python[3531]:     await self.app(scope, receive, send)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/starlette/routing.py", line 65, in app
Oct 14 08:04:03 debian python[3531]:     response = await func(request)
Oct 14 08:04:03 debian python[3531]:   File "/usr/local/lib/python3.9/dist-packages/fastapi/routing.py", line 231, in app
Oct 14 08:04:03 debian python[3531]:     raw_response = await run_endpoint_function(
[... cut ...]
Oct 14 08:04:03 debian python[3531]:     return await dependant.call(**values)
Oct 14 08:04:03 debian python[3531]:   File "./app/routers/setup.py", line 31, in get_status
Oct 14 08:04:03 debian python[3531]:     setupPhase = await redis_get("setupPhase")
Oct 14 08:04:03 debian python[3531]:   File "./app/utils.py", line 64, in redis_get
Oct 14 08:04:03 debian python[3531]:     return v.decode("utf-8")
Oct 14 08:04:03 debian python[3531]: AttributeError: 'str' object has no attribute 'decode'
```

fusion44, 2022-10-16:
> Clean Proxmox 1.8 CLN install:
> 
> ```
> Oct 14 08:04:03 debian python[3531]: ERROR:uvicorn.error:Exception in ASGI application
> Oct 14 08:04:03 debian python[3531]: Traceback (most recent call last):
> line 78, in __call__
> ...
> Oct 14 08:04:03 debian python[3531]:     return await dependant.call(**values)
> Oct 14 08:04:03 debian python[3531]:   File "./app/routers/setup.py", line 31, in get_status
> Oct 14 08:04:03 debian python[3531]:     setupPhase = await redis_get("setupPhase")
> Oct 14 08:04:03 debian python[3531]:   File "./app/utils.py", line 64, in redis_get
> Oct 14 08:04:03 debian python[3531]:     return v.decode("utf-8")
> Oct 14 08:04:03 debian python[3531]: AttributeError: 'str' object has no attribute 'decode'
> ```

Hmm this looks unrelated to the original SSE data connection issue. Was there a warning (similar to `Key 'xxx' not found in Redis DB.` printed in the logs before this error message?

benjamin-wilson, 2022-10-17:
No mentions of `Redis` in the logs

ChuckNorrison, 2022-10-18:
@fusion44 to reproduce please uninstall the web ui + api from blitz menu and install again.

ChuckNorrison, 2022-10-18:
This will fix the issue
https://github.com/fusion44/blitz_api/pull/159

closed 2022-12-15 by rootzoll