raspiblitz/raspiblitz #3263: lnbits not working after installation

rene78, 2022-07-30:
Installed lnbits via the webinterface. Get the following error message when opening the app:
[image]

rene78, 2022-07-30:
Could fix the issue by going to [CONNECT -> RESET.](https://github.com/rootzoll/raspiblitz/issues/3007#issuecomment-1130171980)
Thanks @nicklyk

rootzoll, 2022-08-03:
Check for v1.8.1 also with new poetry install if its working OK.

laurentdebricon, 2022-10-09:
i had the same issue, connect -> reset fixed it.
be aware, your blitz will restart.
i had minor errors with connect --> reset
```
# copy macaroons to central app-data directory and ensure unix ownerships and permissions
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/invoices.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/chainnotifier.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/signer.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/walletkit.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/router.macaroon': No such file or directory
```

closed 2022-12-12 by rootzoll