raspiblitz/raspiblitz #3264: LNbits LNURLp not working

rene78, 2022-07-31:
I have set up a LNURLp link in LNbits but somehow I am not able to send any funds to this link. Tried to send with BlueWallet, Zeus and RaspiBlitz WebUI. The invoice format seems to be wrong.

[image]

v1.8.0
over Tor

rene78, 2022-07-31:
Maybe somehow related to https://github.com/rootzoll/raspiblitz/issues/3129?

rootzoll, 2022-08-03:
@fusion44 can you take a look for the LNbits team whats the problem here?

callebtc, 2022-08-04:
Hey @rene78, please paste the LNURL here so we can check.

rene78, 2022-08-04:
Sure!
LNURL1DP68GURN8GHJ7VFEXGHRZD3C9CCZUDFC8G6NQVP39AKXUATJD3CZ7CTSDYHHVVF0D3H82UNV9UCSFF4YH7
[image]

callebtc, 2022-08-04:
The LNURL [decodes](https://lnurl.fiatjaf.com/codec/) to a local address which can't be reached from outside: `https://192.168.0.58:5001/lnurlp/api/v1/lnurl/1`

How was this usually solved with the RaspiBlitz @rootzoll or @fusion44? Should this be an onion address?

@rene78 do you use any DNS service to reach your node from the internet?

rene78, 2022-08-04:
> @rene78 do you use any DNS service to reach your node from the internet?

No DNS, I use TOR to access stuff like Ride The Lightning.

rootzoll, 2022-08-10:
> How was this usually solved with the RaspiBlitz @rootzoll or @fusion44? Should this be an onion address?

This is normally solved by adding a IP2Tor service with LNbits - so that it can be a Letsencrypt-HTTPS reachable from the outside. To retest & improve those options will get more focus again with v1.8.2 release.

ChuckNorrison, 2022-10-09:
Found this error in my lnbits journal if i try to create a new lnurlp link

> sqlite3.OperationalError: table lnurlp.pay_links has no column named fiat_base_multiplier

dni, 2022-10-25:
hello :wave: @rene78 did you figure it out?

Dehumanizer77, 2023-03-27:
I have the same problem. I have my node running on an internal network and I have a reverse proxy set up on my router which listens on clearnet IP and proxies it to lnbits on raspiblitz. However when creating a LNURL, it encodes the internal IP address into the LNURL, thus making the lnurl invalid for the outside world. If there was a way how to specify the base URL for lnurl, it would solve the problem, but I'm not sure if that is possible...

Dehumanizer77, 2023-03-27:
Ok, for anyone with the similar problem, just use 
```
  RequestHeader set "X-Forwarded-Proto" expr=%{REQUEST_SCHEME}
  RequestHeader set "X-Forwarded-SSL" expr=%{HTTPS}
  ProxyPreserveHost On
``` 
in the reverse proxy config for apache, or
```
    proxy_set_header Host $host;
    proxy_set_header X-Forwarded-Host $host;
    proxy_set_header X-Forwarded-Proto $scheme;
```
for nginx. 
Then it works.

rootzoll, 2024-04-23:
@fusion44 check if still relevent

geco91, 2024-06-03:
recommend to push to v1.11.2

rootzoll, 2024-10-01:
Adding nginx headers adjustments for RaspiBlitz v1.11.3 - will retest in later release when focus on networking again.

rootzoll, 2024-10-02:
Had to remove nginx headers adjustments for RaspiBlitz v1.11.3 again because they made problems:

```
lnbits_ssl.conf
       # Set headers to preserve original scheme and host
        proxy_set_header Host $host;
        proxy_set_header X-Forwarded-Host $host;
        proxy_set_header X-Forwarded-Proto $scheme;
        proxy_set_header X-Forwarded-For $remote_addr;
        proxy_set_header X-Real-IP $remote_addr;
```

To be revisited later again.