raspiblitz/raspiblitz #3265: Bitcoin Core not starting // Block

jessehodl, 2022-07-31:
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 01:47:22 up 13 min,  3 users,  load average: 1.71, 1.85, 1.39

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: deactivating (stop-sigterm) (Result: exit-code)
    Process: 92724 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 92736 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 92738 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=1/FAILURE)
      Tasks: 8 (limit: 4531)
        CPU: 12.336s
     CGroup: /system.slice/bitcoind.service
             └─92748 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Jul 29 01:47:18 raspberrypi bitcoind[92738]: Error during initializaton - check debug.log for details
Jul 29 01:47:18 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Thu 2022-07-28 15:24:55 BST, ends at Fri 2022-07-29 01:47:22 BST. --
Jul 29 01:47:05 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
Jul 29 01:47:05 raspberrypi systemd[1]: bitcoind.service: Consumed 13.761s CPU time.
Jul 29 01:47:06 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 22.
Jul 29 01:47:06 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Jul 29 01:47:06 raspberrypi systemd[1]: bitcoind.service: Consumed 13.761s CPU time.
Jul 29 01:47:06 raspberrypi systemd[1]: Starting Bitcoin daemon...
Jul 29 01:47:18 raspberrypi bitcoind[92738]: Error during initializaton - check debug.log for details
Jul 29 01:47:18 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-07-29T00:47:06Z * Using 55.8 MiB for basic block filter index database
2022-07-29T00:47:06Z * Using 8.0 MiB for chain state database
2022-07-29T00:47:06Z * Using 382.5 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2022-07-29T00:47:06Z init message: Loading block index…
2022-07-29T00:47:06Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2022-07-29T00:47:06Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2022-07-29T00:47:06Z Opened LevelDB successfully
2022-07-29T00:47:06Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2022-07-29T00:47:17Z LoadBlockIndexDB: last block file = 3135
2022-07-29T00:47:17Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=51, size=56638164, heights=747128...747178, time=2022-07-29...2022-07-30)
2022-07-29T00:47:17Z Checking all blk files are present...
2022-07-29T00:47:18Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2022-07-29T00:47:18Z Opened LevelDB successfully
2022-07-29T00:47:18Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: 032202509162ec84
2022-07-29T00:47:18Z Loaded best chain: hashBestChain=00000000000000000004c831db6f816b56061ca94c24c88a02807ecd7490673a height=747122 date=2022-07-29T20:26:39Z progress=1.000000
2022-07-29T00:47:18Z init message: Verifying blocks…
2022-07-29T00:47:18Z : The block database contains a block which appears to be from the future. This may be due to your computer's date and time being set incorrectly. Only rebuild the block database if you are sure that your computer's date and time are correct.
Please restart with -reindex or -reindex-chainstate to recover.
2022-07-29T00:47:18Z Aborted block database rebuild. Exiting.
2022-07-29T00:47:18Z Shutdown: In progress...

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

[... 412 lines cut ...]
ln_default_activated="1"
btc_default_online="0"
btc_default_activated="1"
btc_default_ready="0"
btc_default_running="0"
btc_default_error_full=""
btc_default_error_short=""
btc_default_peers="0"
btc_default_port=""
btc_default_sync_percentage="0"
btc_default_sync_initial_done="0"
btc_default_version="v23.0.0"

*** LOGFILES ***
Archived and active journals take up 392.0M in the file system.
430M    /var/log

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

[image]

22388o, 2022-08-01:
You tried make this: https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#why-is-my-final-sync-taking-so-long ? And after reboot.

rootzoll, 2022-08-03:
The problem is this lines in the debug:
```
2022-07-29T00:47:18Z init message: Verifying blocks…
2022-07-29T00:47:18Z : The block database contains a block which appears to be from the future. This may be due to your computer's date and time being set incorrectly. Only rebuild the block database if you are sure that your computer's date and time are correct.
Please restart with -reindex or -reindex-chainstate to recover.
2022-07-29T00:47:18Z Aborted block database rebuild. Exiting.
2022-07-29T00:47:18Z Shutdown: In progress...
```

@jessehodl try ro rebuild the index like shown here: https://github.com/rootzoll/raspiblitz/issues/3016#issuecomment-1083187550

cstenglein, 2023-06-07:
@jessehodl was this issue resolved?

Closing for now, if it still doesn't work, please re-open