raspiblitz/raspiblitz #3267: Upgrade Not working with error message below

pkepler, 2022-07-31:
can someone help me?  I have done two fresh image burns trying to upgrade.  I am able to change Password A and then this come su::

# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.8.0'
lcdrotate='1'
lightning=''
network='bitcoin'
chain='main'
hostname=''
runBehindTor='on'
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
# setting PASSWORD A
# salt file exists
# hashedPasswordSalt(12e7fc37835ee05d)
runningUser='0'
# OK - password A changed for user pi, root, admin & bitcoin
error=''
Calling _provision.update.sh ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-update.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(missing-hostname)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
admin@192.168.1.79:~ ₿

22388o, 2022-08-01:
You tried make reboot? You verified your logs?

pkepler, 2022-08-01:
I did a reboot, I have gone ahead and re-burned a 3rd fresh image.  What does verify my logs mean?

pkepler, 2022-08-01:
I completed the third image burn and started the Raspiblitz up.  I found a configuration on my SDD and starts the recovery process.  I input a new Password A and then it returns the error  "state(error) message(missing-hostname)"

Upon reboot it directs me to "PLEASE START WITH A FRESH SD CARD IMAGE"

22388o, 2022-08-01:
> I did a reboot, I have gone ahead and re-burned a 3rd fresh image. What does verify my logs mean?

It's a repository of all information related to booting and any messages logged.

22388o, 2022-08-01:
> I completed the third image burn and started the Raspiblitz up. I found a configuration on my SDD and starts the recovery process. I input a new Password A and then it returns the error "state(error) message(missing-hostname)"
> 
> Upon reboot it directs me to "PLEASE START WITH A FRESH SD CARD IMAGE"

You're connected to the Tor?

rootzoll, 2022-08-03:
@pkepler the error comes from this line in the update script: https://github.com/rootzoll/raspiblitz/blob/b20293a0c90f531477774c9a810b5491c81d0166/home.admin/_provision.update.sh#L51

So n your RaspiBlitz config the value `hostname` is missing and needs to be set so that the next fresh sd card updadte runs thru. You edit the config `/mnt/hdd/raspiblitz.conf` manually or directly from the terminal run:
- if you use LND: `/home/admin/config.scripts/lnd.setname.sh myhostname`
- if you use Core Lightning: `/home/admin/config.scripts/lnd.setname.sh myhostname`

rootzoll, 2022-08-03:
for 1.8.1 the script should auto-repair and set `raspberry` as an repair value

henkisdabro, 2023-10-06:
Unsure whether this is unrelated or not, but posting that I ran into a similar issue today when attempting to upgrade from 1.9.0 to 1.10.0. I downloaded and flashed the "FATPACK" file and verified it against the SHA256 hash first. I'm unsure as to what my next step is. 

Here is what's showing in the terminal window:

```
###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.10.0
Fri  6 Oct 09:06:53 BST 2023
***********************************************
# *** /home/admin/config.scripts/blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
## INIT raspiblitz.info
baseimage=raspios_arm64
cpu=aarch64
blitzapi=on
displayClass=lcd
displayType=
setupPhase=boot
setupStep=0
fsexpanded=0
state=starting
btc_mainnet_sync_initial_done=0
btc_testnet_sync_initial_done=0
btc_signet_sync_initial_done=0
ln_lnd_mainnet_sync_initial_done=0
ln_lnd_testnet_sync_initial_done=0
ln_lnd_signet_sync_initial_done=0
ln_cl_mainnet_sync_initial_done=0
ln_cl_testnet_sync_initial_done=0
ln_cl_signet_sync_initial_done=0
## INIT RaspiBlitz Cache ... wait background.scan.service to finish first scan loop
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime(8)
OK - logs are at 50 MB - within safety limit
Waiting for HDD/SSD ...
total 612
drwxr-xr-x   2 root root   4096 Oct  6 09:06 .
drwxr-xr-x 127 root root  12288 Oct  6 09:06 ..
-rw-r--r--   1 root root 565189 Dec  2  2021 moduli
-rw-r--r--   1 root root   1580 Dec  2  2021 ssh_config
-rw-r--r--   1 root root   3250 Dec  2  2021 sshd_config
-rw-------   1 root root   1381 Dec  2  2021 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Dec  2  2021 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Dec  2  2021 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Dec  2  2021 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Dec  2  2021 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Dec  2  2021 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Dec  2  2021 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Dec  2  2021 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Dec  2  2021 ssh_import_id
HDD/SSD connected: sda
No HDMI switch found.
No FS EXPAND needed. needsExpansion(0) fsexpanded(0)
No WIFI RESTORE because no /var/cache/raspiblitz/hdd-inspect/wpa_supplicant.conf
SSH SERVER CERTS RESTORE activating old SSH host keys
# *** /home/admin/config.scripts/blitz.ssh.sh restore
# restore sshd host keys from: /var/cache/raspiblitz/hdd-inspect/sshd/ssh/sshd
# OK - sshd keys restore done
[... 123 lines cut ...]
mmcblk0
├─mmcblk0p1 vfat   bootfs
└─mmcblk0p2 ext4   rootfs
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
# migrationFile()
# Sourcing /mnt/hdd/raspiblitz.conf
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
# setting PASSWORD A
# creating salt & hashedPasswordStoragePath ...
# hashedPasswordSalt()
runningUser='0'
error='hdd not mounted yet - cannot set/check blitz passwords yet'
correct=0
Provisioning  Mainnet - not active
Provisioning  Testnet - not active
Provisioning  Signet - not active
Calling _provision.update.sh ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-update.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(missing-hostname)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
```

rootzoll, 2023-10-18:
TODO for devs: next release auto-repair and set raspberry as an repair value here: https://github.com/rootzoll/raspiblitz/blob/b20293a0c90f531477774c9a810b5491c81d0166/home.admin/_provision.update.sh#L51

PR #4272 (2023-11-23): Repair Hostname
fix #3267 : https://github.com/raspiblitz/raspiblitz/issues/3267#issuecomment-1769048708
closed 2023-11-23 by rootzoll