raspiblitz/raspiblitz #3270: New web interface "Transactions" error

davidepatti, 2022-08-01:
[image]

Everythings else seems just fine.

redmac68, 2022-08-01:
+1
[image]

rootzoll, 2022-08-03:
👀 @fusion44

fusion44, 2022-08-03:
Looks like a gRPC error which might originate from the gRPC server.

@davidepatti @redmac68 Do you run CLN or LND? How many transactions do you roughly have?

redmac68, 2022-08-03:
I run LND.
In July 2255 transactions.

davidepatti, 2022-08-03:
> Looks like a gRPC error which might originate from the gRPC server. 
> @davidepatti @redmac68 Do you run CLN or LND? How many transactions do you roughly have?

I use LND, since december 2021, usually like 200/300 transactions per month

cstenglein, 2023-06-07:
@davidepatti @redmac68  Does this still occur?

Sebe0401, 2023-08-30:
Having the same problem... is there a solution already?

rootzoll, 2024-04-23:
@Sebe0401 do you still have this issue with v1.11.0?

Sebe0401, 2024-04-26:
> @Sebe0401 do you still have this issue with v1.11.0?

@rootzoll yes, it only shows a rotating circle and then this message pops up. 
[image]

Maybe i have too many transactions? I am doing many rebalance requests. These requests normally showed up there.

rootzoll, 2024-05-25:
@fusion44 can you take a look again?

fusion44, 2024-05-26:
I will, but I'm not certain that there is a good solution, except upstream letting me query for only a portion of the data at a time. Maybe things have changed since I've last looked.