raspiblitz/raspiblitz #3271: createwallet command

jaonoctus, 2022-08-03:
Why `createwallet` is not available?

Tried to run `bitcoin-cli createwallet ...` but got:

```
error code: -32601
error message:
Method not found
```

`$ bitcoin-cli help | grep create`

```
createpsbt ...
createrawtransaction ...
createmultisig ...
```

rootzoll, 2022-08-03:
By default in the `bitcoin.conf` wallet is active. If you need it you can change your config under `/mnt/hdd/bitcoin/bitcoin.conf` - but make sure you take care yourself of backuping the funds in that wallet.

jaonoctus, 2022-08-03:
You meant inactive by default?

https://github.com/rootzoll/raspiblitz/blob/b20293a0c90f531477774c9a810b5491c81d0166/home.admin/assets/bitcoin.conf#L11

So basically change `disablewallet=1` to `disablewallet=0` 👍🏻 

Thank you

closed 2022-08-03 by jaonoctus