raspiblitz/raspiblitz #3272: Updated API version ready?

rootzoll, 2022-08-03:
Plan is to deliver with API version 0.6.0
https://github.com/fusion44/blitz_api/milestone/2

rootzoll, 2022-12-15:
We agreed on having similar branch namings for releases (v1.8, v1.9, v1.10, ..) also in API & WebUI repos like on RaspiBlitz repo to keep releases patchable.

@fusion44 please create a `v1.9` branch on blitz_api repo merged will all teh code you have ready for the next release. I will adapt the api install script on the raspiblitz side to try to use such release branches.

rootzoll, 2022-12-19:
as soon as `refactor_system` is renamed to `dev` and `blitz-v1.9` is available - merge: https://github.com/rootzoll/raspiblitz/pull/3546

rootzoll, 2022-12-22:
To be fixed for rc2:
- [ ] https://github.com/fusion44/blitz_api/issues/184

cstenglein, 2023-06-07:
Should be done => closed

closed 2023-06-07 by cstenglein