raspiblitz/raspiblitz #3273: Updated WebUI ready?

rootzoll, 2022-08-03:
Plan is to deliver with at least v1.0.1
https://github.com/cstenglein/raspiblitz-web/milestone/3

rootzoll, 2022-12-15:
We agreed on having similar branch namings for releases (v1.8, v1.9, v1.10, ..) also in API & WebUI repos like on RaspiBlitz repo to keep releases patchable.

@cstenglein please create a v1.9 branch on raspiblitz-web repo merged will all the code you have ready for the next release. I will adapt the webui install script on the raspiblitz side to use such release branches.

cstenglein, 2022-12-16:
Done: `release/v1.9`

See https://github.com/cstenglein/raspiblitz-web/branches

Regarding merges: Some PRs still require functionality from the API like https://github.com/cstenglein/raspiblitz-web/pull/527
Should I merge that as well?

closed 2023-02-03 by rootzoll