raspiblitz/raspiblitz #3278: Optimise the update process

P1xelfehler, 2022-08-08:
As an owner of a lightning node, I want to be able to have as much uptime as possible.

Currently when I insert the SD card with the new image, all the services are being installed before the lightning node itself starts.

I think it would be much much better if we first get the lightning node itself up and running before we setup all the other apps and services to be up and running asap for keeping the coins flowing through our node.

rootzoll, 2022-08-10:
I can see that uptime is important, but the raspberry is quite limited in processing power .. running bitcoin core (maybe even with some blocks to validate) and the also lightning during all the other setup processes are happening seems a bit much for that system. Maybe when the way we install apps gets changed we can take another look at this - but this would be quite some time in future.

openoms, 2022-08-19:
Can speed up the process by manually deactivating the services in the
`/mnt/hdd/raspiblitz.conf`
so they won't get set up during recovery. This way could get the node running ASAP then install from SERVICES again.
To deactivate non-essential services befora update could be an option in the menu.

P1xelfehler, 2022-08-19:
That’s a great idea 👍