raspiblitz/raspiblitz #3279: Updating RaspiBlitz to new Version (Bitcoin Core RPC)

btcpuertorico, 2022-08-08:
https://github.com/rootzoll/raspiblitz#updating-raspiblitz-to-new-version

_"It is worth noting here that your SSH password will be reset to raspiblitz."_

Also noticed that Bitcoin Core RPC needs to be disabled and re-enabled for it to accept connections again. All clients (Sparrow/Specter) lost connectivity.

rootzoll, 2022-08-10:
Can you describe more in detail what you mean here? Basically on an update a complete new bitcoin core gets installed with reboots and everything - that clients loose connectivity would be expected if open.

btcpuertorico, 2022-08-10:
I mean, seems like bitcoin RPC service stopped listening after reboot. No authentication error, just no response.

rootzoll, 2022-08-21:
Can you spot any error message regarding the RPC startup when you check the bitcoin logs after reboot?

closed 2023-06-07 by cstenglein