raspiblitz/raspiblitz #3281: WebUI: Add Update & Recovery of old node/backup on Setup

rootzoll, 2022-08-10:
I think this is one of the important things to add for the future - that users get assisted thru the update process and get offered to download the lnd/cl-rescue backup at one point. And during Setup a rescue-file can be used to recover an old node.

@cstenglein how do we wanna start the design/frontend part of this? Should I create some mockup first we then can discuss?

cstenglein, 2022-08-14:
I think it could be solved in one screen for the backup & one to download on recover.

Not sure about where in the flow it should be placed though.

rootzoll, 2022-08-21:
> Not sure about where in the flow it should be placed though.

np. will prepare a mockup once we will pickup this issue for v1.8.1

YBHQ, 2022-08-22:
~~~~

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

YBHQ, 2022-08-22:
#

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

rootzoll, 2022-12-15:
We will push this to v1.10 because if time constraints.

cstenglein, 2024-07-27:
Is there an idea / mockup how to integrate it?

And is there an API endpoint to download a backup @fusion44?

rootzoll, 2024-07-29:
> Is there an idea / mockup how to integrate it?

Not yet - would it make sense to plan/do this together with the general redesign of the dashboard?