raspiblitz/raspiblitz #3283: CPU temps don't change in amd64

wchancao, 2022-08-11:
The CPU temperatures shown in 00infoBlitz.sh don't change in an amd64 Dell Optiplex 5040, they always stay at 27C and 81F. They work in an amd64 Dell Optiplex 7010. I can get the temps installing lmsensors and running sensors.

wchancao, 2022-08-12:
Looking through the scripts and googling a little bit, it seems that the correct temp for the cpu is the one displayed as `x86_pkg_temp` in `cat /sys/class/thermal/thermal_zone*/type`
Right now, the script in Raspiblitz is pointing to `/sys/class/thermal/thermal_zone0/temp` which is `acpitz`

BitromNO, 2022-08-15:
have the same issue on MSI Cubi-N

openoms, 2022-08-15:
not sure how to provide a solution which works on all hardware, but `lm-sensors` is a good one.
On my Debian laptop:
```
₿ sudo apt install lm-sensors
₿ sensors
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +43.5°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +46.0°C  (high = +100.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +44.0°C  (crit = +128.0°C)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:        3229 RPM
temp1:        +44.0°C  
temp2:            N/A  
temp3:         +0.0°C  
temp4:         +0.0°C  
temp5:         +0.0°C  
temp6:         +0.0°C  
temp7:         +0.0°C  
temp8:         +0.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          17.18 V  
```

openoms, 2022-08-15:
an other good option is chechingthe temperature in `htop`:
* run `htop`
* press F2
* Display options
*  [x]    Also show CPU temperature (requires libsensors)

BitromNO, 2022-08-17:
Ive been using htop for a while. 
What I want is to identify what temp sensor I need, and change it in the script or if there is a conf file or something, so it shows in the info screen.

rootzoll, 2022-12-11:
@openoms can this issue be closed?

openoms, 2022-12-11:
Not actionable for now. If someone has a solution for the script it is welcome.

closed 2022-12-11 by openoms