raspiblitz/raspiblitz #3285: v1.8 build_sdcard.sh: Adding deb.torproject.org keyring failed

adevoss, 2022-08-13:
When updating to v1.8 (amd64 VM) the `build_sdcard.sh` failed.

First I clean installed Debian 11 then executed `build_sdcard.sh`
```
For details on optional parameters - call with '--help' or check source code.
interaction=true
fatpack=true
github_user=rootzoll
branch=v1.8
display=headless
tweak_boot_drive=false
wifi_region=off
cpu=x86_64
architecture=amd64
baseimage=debian
```
```
*** Adding deb.torproject.org keyring ***
gpg: no valid OpenPGP data found.
! FAIL: Was not able to import deb.torproject.org key
```
I think the issue is in `/home/admin/config.scripts/tor.install.sh` line 47
`add_tor_sources()`

Unfortunately I don't know how to solve this.

adevoss, 2022-08-14:
This completed the build_sdcard.sh script. If this is a workaround or a solution I don't know. Please comment.
```
sudo wget -qO- https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | sudo gpg --dearmor | sudo tee /usr/share/keyrings/tor-archive-keyring.gpg >/dev/null
sudo apt update
sudo apt install tor deb.torproject.org-keyring
```

openoms, 2022-08-15:
Needs a clearnet fallback: https://github.com/rootzoll/raspiblitz/issues/3249

openoms, 2022-08-15:
Please check if 35607331 fixed this for you.

adevoss, 2022-08-16:
Short version: Yes.

https://github.com/rootzoll/raspiblitz/issues/3285#issuecomment-1214419645 is similar to https://github.com/rootzoll/raspiblitz/commit/35607331de89df6d92b9ae3d19c8c4d005345425 so it was already 'fixed' for me.
Tor didn't time-out now.
So I added your code to the `tor.install.sh` script manually and forced a clear net update by modifying the `if` statement.
Then I executed `sudo tor.install.sh update`.
That got me an update over clearnet.
Then I changed back to the original `if` statement.

openoms, 2022-08-16:
Thanks for testing.

closed 2022-08-16 by openoms