raspiblitz/raspiblitz #3288: [v1.8] RTL is 'on' in menu but is an available app in web interface

adevoss, 2022-08-15:
Installing RTL might have gone wrong but it shouldn't be shown as installed in the menu when you connect via ssh and be an available app in the web interface.

adevoss, 2022-08-15:
I reinstalled RTL in previous versions via the menu so now I tried to reinstalled it via the webinterface.
Successfully.

rootzoll, 2022-08-21:
TODO: Add for all apps that are available via the WebUI a final install check before setting `on` in config.

rootzoll, 2022-12-19:
App instals have now a fatpack option - that sjhould fix most oif install problems for regulat users. But sctipts to be improved in the future - so pushing this to v1.10

cstenglein, 2023-06-07:
@adevoss Can you check again on 1.9.0? Should be fixed AFAIK.

Closing this for now, if it still happens, please re-open.

closed 2023-06-07 by cstenglein

adevoss, 2023-06-11:
:ok_hand: