raspiblitz/raspiblitz #3289: [v1.8] Helipad webinterface not shown

adevoss, 2022-08-15:
When Helipad is installed via the 'ssh menu' the webinterface is not shown in a browser.
Port 2112 is not shown when executing `sudo netstat -tulpn|grep 2112` on the raspiblitz.
Port 2113 is shown when executing `sudo netstat -tulpn|grep 2113` on the raspiblitz.
On port 2113 https://<localip>:2113: 502 Bad Gateway

adevoss, 2022-08-15:
For nginx there is no configuration file for Helipad using port 2112.
When helipad-ssl.conf is copied to helipad.conf and port 2113 ssl changed to 2112 the logs are:
```
... [error] ..... connect() failed (111: Connection refused) while connecting to upstream, client: .... request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:2112/", host: "<raspiblitzip>:2113"
... [error] ..... connect() failed (111: Connection refused) while connecting to upstream, client: .... request: "GET /favicon.ico HTTP/1.1", upstream: "http://127.0.0.1:2112/favicon.ico", host: "<raspiblitzip>:2113", referrer: "https://<raspiblitzip>:2113/"
... [error] ..... connect() failed (111: Connection refused) while connecting to upstream, client: .... request: "GET / HTTP/1.1", upstream: "http://127.0.0.1:2112/", host: "192.168.3.205:2113"
... [alert] ..... worker_connections are not enough while connecting to upstream
... [alert] ..... worker_connections are not enough while connecting to upstream, client: 127.0.0.1, server: _, request: "GET / HTTP/1.0", upstream: "http://127.0.0.1:2112/", host: "127.0.0.1:2112"
```
`nginx -t` is OK and nginx is restarted.

Must not the helipad program listen on the ports 2112 and 2113 instead of nginx?

adevoss, 2022-08-15:
The problem were not the nginx config files.
The solution (for amd64) is to install an extra package: pkg-config.

bonus.helipad.sh line 93:
`sudo apt --assume-yes --show-upgraded install libssl-dev libsqlite3-dev pkg-config`

rootzoll, 2022-08-21:
thx. PR #3290 marked for merge

closed 2022-12-11 by rootzoll