raspiblitz/raspiblitz #3291: state(error) message(lnd-wallet-seed)

Ratzefummel4, 2022-08-15:
Hello guys,
I had setup a new lightning node and want to migrate my old wallet from umbrel. I have been stucking now with this error:

 `# df 
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30367292 9960812  19137324  35% /
devtmpfs         1933668       0   1933668   0% /dev
tmpfs            1968324      16   1968308   1% /dev/shm
tmpfs             787332    1012    786320   1% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs              32768       8     32760   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095   30830    227266  12% /boot
tmpfs             393664      16    393648   1% /run/user/1000
tmpfs             393664      16    393648   1% /run/user/1001
/dev/sda1      960302804     112 911448232   1% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   BLOCKCHAIN
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# migrationFile()
# Sourcing /mnt/hdd/raspiblitz.conf 
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.8.0'
lcdrotate='1'
lightning='lnd'
network='bitcoin'
chain='main'
hostname='Ratzefummel'
runBehindTor='on'
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# setting PASSWORD A
# creating salt & hashedPasswordStoragePath ...
# hashedPasswordSalt(fa9820f4b1fa8c24)
runningUser='0'
# OK - password A changed for user pi, root, admin & bitcoin
error=''
Calling _provision.setup.sh for basic setup tasks ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-setup.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(lnd-wallet-seed)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
`
Probably I did a mistake with the seed words and I have to restart the setup process? Could you help please? I'm completely new in the raspiblitz space. thanks

openoms, 2022-08-16:
the way it should be that you flash the SDcard image of the raspiblitz and place it in the umbrel node. That's how the migration is taking place. Is that what you are doing?

Christian891, 2022-08-16:
Hello openoms,
I have got a second RasberrryPI for new setup. While the setup is being in process the umbrel node should still running therefor I setup a second RasberryPi.
I came now to the decision that I will flash the SD-Card again and start the setup from the very beginning. Thanks for your reply

cstenglein, 2023-06-07:
Seems to be resolved => closed