raspiblitz/raspiblitz #3295: CLN: make the wallet reset more reliable

openoms, 2022-08-18:
the old config and channel database can be left
behind when using SEEDRESTORE and FILERESTORE

openoms, 2022-08-18:
While testing found that

https://github.com/rootzoll/raspiblitz/blob/7074455380eec200bfcaa35cd6cca066fcf9bef3/home.admin/99clRepairMenu.sh#L195

does not delete the files in the directory for an unknown reason to me.
After #3296 the while .lightning/bitcoin directory is deleted and then recreated so the cleanup works.

This behaviour is possibly related to: https://github.com/ElementsProject/lightning/issues/5119

openoms, 2022-08-18:
@21OrangePill and anyone using `SEEDRESTORE` or `FILERESTORE` on a running CLN node please run `patch` on `v1.8`

YBHQ, 2022-08-22:
~~~~

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

openoms, 2022-08-26:
closing as tested successfully, Will be part of v1.8.1

PR #3296 (2022-08-18): fix CLN wallet reset
fixes: #3295
closed 2022-08-26 by openoms