raspiblitz/raspiblitz #3297: [v1.8] Unable to install RTL

Tracachang, 2022-08-18:
I am unable to install RTL for c-lightning. I'm asked to reboot and try manually with the script bonus.rtl.sh on but I do get the same error. 

Log:

>  npm resolution error report
> 
> While resolving: @angular-eslint/builder@12.2.1
> Found: @angular/cli@13.1.1
> node_modules/@angular/cli
>   peer @angular/cli@">= 13.0.0 < 14.0.0" from @angular-eslint/schematics@13.0.1
>   node_modules/@angular-eslint/schematics
>     dev @angular-eslint/schematics@"13.0.1" from the root project
>   dev @angular/cli@"^13.0.3" from the root project
> 
> Could not resolve dependency:
> peer @angular/cli@">= 12.0.0 < 13.0.0" from @angular-eslint/builder@12.2.1
> node_modules/@angular-eslint/builder
>   dev @angular-eslint/builder@"^12.2.1" from the root project
> 
> Conflicting peer dependency: @angular/cli@12.2.18
> node_modules/@angular/cli
>   peer @angular/cli@">= 12.0.0 < 13.0.0" from @angular-eslint/builder@12.2.1
>   node_modules/@angular-eslint/builder
>     dev @angular-eslint/builder@"^12.2.1" from the root project
> 
> Fix the upstream dependency conflict, or retry
> this command with --force, or --legacy-peer-deps
> to accept an incorrect (and potentially broken) dependency resolution.
> 
> Raw JSON explanation object:

> {
>   "code": "ERESOLVE",
>   "edge": {
>     "type": "peer",
>     "name": "@angular/cli",
>     "spec": ">= 12.0.0 < 13.0.0",
>     "error": "INVALID",
>     "from": {
>       "name": "@angular-eslint/builder",
>       "version": "12.2.1",
>       "location": "node_modules/@angular-eslint/builder",
>       "isWorkspace": false,
>       "dependents": [
>         {
>           "type": "dev",
>           "name": "@angular-eslint/builder",
>           "spec": "^12.2.1",
>           "from": {
>             "location": "/home/rtl/RTL"
>           }
>         }
>       ]
>     }
>   },
>   "dep": {
>     "name": "@angular/cli",
>     "version": "12.2.18",
>     "whileInstalling": {
>       "name": "@angular-eslint/builder",
>       "version": "12.2.1",
>       "path": "/home/rtl/RTL/node_modules/@angular-eslint/builder"
>     },
>     "location": "node_modules/@angular/cli",
>     "isWorkspace": false,
>     "dependents": [
>       {
>         "type": "peer",
>         "name": "@angular/cli",
>         "spec": ">= 12.0.0 < 13.0.0",
>         "error": "INVALID",
>         "from": {
>           "name": "@angular-eslint/builder",
>           "version": "12.2.1",
>           "location": "node_modules/@angular-eslint/builder",
>           "isWorkspace": false,
>           "dependents": [
>             {
>               "type": "dev",
>               "name": "@angular-eslint/builder",
>               "spec": "^12.2.1",
>               "from": {
>                 "location": "/home/rtl/RTL"
>               }
>             }
>           ]
>         }
>       }
>     ]
>   },
>   "current": {
>     "name": "@angular/cli",
>     "version": "13.1.1",
>     "location": "node_modules/@angular/cli",
>     "isWorkspace": false,
>     "dependents": [
>       {
>         "type": "peer",
>         "name": "@angular/cli",
>         "spec": ">= 13.0.0 < 14.0.0",
>         "from": {
>           "name": "@angular-eslint/schematics",
>           "version": "13.0.1",
>           "location": "node_modules/@angular-eslint/schematics",
>           "isWorkspace": false,
>           "dependents": [
>             {
>               "type": "dev",
>               "name": "@angular-eslint/schematics",
>               "spec": "13.0.1",
>               "from": {
>                 "location": "/home/rtl/RTL"
>               }
>             }
>           ]
>         }
>       },
>       {
>         "type": "dev",
>         "name": "@angular/cli",
>         "spec": "^13.0.3",
>         "from": {
>           "location": "/home/rtl/RTL"
>         }
>       }
>     ]
>   },
>   "peerConflict": {
>     "current": {
>       "name": "@angular/cli",
>       "version": "13.1.1",
>       "location": "node_modules/@angular/cli",
>       "isWorkspace": false,
>       "dependents": [
>         {
>           "type": "peer",
>           "name": "@angular/cli",
>           "spec": ">= 13.0.0 < 14.0.0",
>           "from": {
>             "name": "@angular-eslint/schematics",
>             "version": "13.0.1",
>             "location": "node_modules/@angular-eslint/schematics",
>             "isWorkspace": false,
>             "dependents": [
>               {
>                 "type": "dev",
>                 "name": "@angular-eslint/schematics",
>                 "spec": "13.0.1",
>                 "from": {
>                   "location": "/home/rtl/RTL"
>                 }
>               }
>             ]
>           }
>         },
>         {
>           "type": "dev",
>           "name": "@angular/cli",
>           "spec": "^13.0.3",
>           "from": {
>             "location": "/home/rtl/RTL"
>           }
>         }
>       ]
>     },
>     "peer": {
>       "name": "@angular/cli",
>       "version": "12.2.18",
>       "whileInstalling": {
>         "name": "@angular-eslint/builder",
>         "version": "12.2.1",
>         "path": "/home/rtl/RTL/node_modules/@angular-eslint/builder"
>       },
>       "location": "node_modules/@angular/cli",
>       "isWorkspace": false,
>       "dependents": [
>         {
>           "type": "peer",
>           "name": "@angular/cli",
>           "spec": ">= 12.0.0 < 13.0.0",
>           "error": "INVALID",
>           "from": {
>             "name": "@angular-eslint/builder",
>             "version": "12.2.1",
>             "location": "node_modules/@angular-eslint/builder",
>             "isWorkspace": false,
>             "dependents": [
>               {
>                 "type": "dev",
>                 "name": "@angular-eslint/builder",
>                 "spec": "^12.2.1",
>                 "from": {
>                   "location": "/home/rtl/RTL"
>                 }
>               }
>             ]
>           }
>         }
>       ]
>     }
>   },
>   "strictPeerDeps": false,
>   "force": false,
>   "isMine": true
> }
>

Tracachang, 2022-08-20:
I've edit the script and add --legacy-peer-deps and it works.

rootzoll, 2022-08-21:
@openoms I havent seen this proble  with RTL install for core lightning during testing. Can you also recommend to add PR #3299 for v1.8.1 to have this more stable in the future?

openoms, 2022-08-21:
@Tracachang can you test: https://github.com/rootzoll/raspiblitz/pull/3293 please

Tracachang, 2022-08-21:
> @Tracachang can you test: #3293 please

Yes, just tested and it works, no need for https://github.com/rootzoll/raspiblitz/pull/3299.

closed 2022-08-21 by Tracachang