raspiblitz/raspiblitz #3300: feature request: using kindle-dash to add screen for RaspiBlitz

woeisme, 2022-08-22:
many of us are running our RaspiBlitz headless but would like to add screens. 
Sometimes acquiring an old kindle is easier than a full screen display.
[This project](https://github.com/pascalw/kindle-dash) repurposes old (jailbroken!) kindles as a display device, having the kindle device itself fetch data via curl at regular intervals. 

I think it would make a nice integration to the RaspiBlitz project.

I have a spare PW kindle I may begin tinkering with and update this ticket as needed

NilByte, 2022-08-22:
Check this one.

https://github.com/rootzoll/raspiblitz/blob/v1.8/home.admin/config.scripts/bonus.kindle-display.sh

openoms, 2022-08-23:
Yes, already using https://github.com/dennisreimann/kindle-display and as above can connect to the blitz.

Modifications to the server (in JS) are welcome to show any other info.