raspiblitz/raspiblitz #3302: webui does not come up again after deinstall/reinstall

jagottsicher, 2022-08-25:
I had a running node 1.7.0 which I updated to 1.8.0.
All worked fine to some point and I had a fully synced node with webui up and running.

For some reason I needed to reinstall everything (at the same day). afterwards the webgui (of the synced node) did not show the correct values anmore, but hung insiting on that bitcoin was off and lnd not synced. So I stiwtched to the menu and deintsalled the webui, restarted and installed the webui again. Now my node is synced, btc and lnd both synced and running, redis in the process list, but my webui shows an error only -- 403 Forbidden, which seems that nginx (which I assume is a reverse proxy only) cannot get to any info provided ny nodejs/redis (or whatever).

I already restarted nginx, redis and rebootet - no change. my btc node and lnd both look fine for me.

How to restart the webui without rewriting the complete sd-card?

cstenglein, 2022-11-10:
Hi @jagottsicher 

Can you try to disable & enable the API & webUI again?

cstenglein, 2023-03-23:
Should work with 1.9.0

closed 2023-05-16 by rootzoll