raspiblitz/raspiblitz #3303: Request: Core Lightning emergencyrecover backup script

exquisiteskink, 2022-08-26:
With CLN 0.12 a new feature was introduced similar to static channel backups in lnd. Requesting a script be added to raspiblitz that will automatically create a new backup with channel changes and save that backup to a remote location similar to how raspiblitz currently backs up the SCB files.

openoms, 2022-08-26:
https://blog.blockstream.com/core-lightning-v0-12-0/
Video demo: https://www.youtube.com/watch?v=_oaJc4ISzTE

after the update there is an `emergency.recover` file in /home/bitcoin/.lightning/bitcoin near the `hsm_secret`

The  `emergency.recover` is encrypted (? with the root key) so safe to share on nextcloud, USB drive etc.

openoms, 2022-11-09:
backup is implemented in #3348

closed 2022-12-15 by rootzoll